Pulse-Eight/libcec #261: CEC is messing with audio source when RPI connected

barat, 2016-10-29:
Hi,
I have a strange at least a year old issue whit is irritating me ...
My setup is:
Philips 55PFS8109/12 TV (EasyLink + ARC enabled) connected to Philips Fidelio B5 + RPI (osmc) connected to B5 as well.
Pi is working 24/7

**Reproduce path:**

1. Turn TV on, see soundbar ON and sound is going through ARC (ARC displayed on B5)
2. Switch source to Raspberry (it's on the source list) - soundbar is changing sound to B5 HDMI 2
3. Everything works, CEC is adjusting volume on B5 and interacting with Kodi on Raspberry
4. Switch back to TV source (DVB-T)

**Expected result:**
Source switched to TV, sound playing through ARC

**What is happening instead:**
Source switched to TV, ARC displayed on soundbar for 1-2s and then again changed to HDMI2. Switching to ARC using B5 remote control again gives ARC for 1-2s (even hear DVB-T sound), and again to HDMI2. 

When I turn TV OFF, both TV and B5 are OFF (good), but sometimes in 2-10s B5 wakes up, turning on TV and changing to B5 HDMI2 (and after a while to Fidelio B5 HDMI1 source with no picture). From there I can switch to TV (DVB-T) and I again have ARC sound ...

Logs (cec-client):
http://pastebin.com/rPYRR6gU

Does someone know what's going on, and if 4.0.0 libcec will handle it?

opdenkamp, 2016-11-02:
I don't see anything in this log that libCEC is doing when the TV's source is activated that would cause this. Is it actually happening in this log?

barat, 2016-11-02:
Yes ... this log was recorded when i was doing steps 1-4 + "What is happening instead".
But it may be possible that it's ok now in 4.0.0, OSMC included it in test build. After clean install I was able to switch sources without issues, but there was no time to add media source and play something.
Will report today about this.

barat, 2016-11-02:
After short test it seems that 4.0.0 with Kodi 17 on OSMC beta is acting properly ... fingers crossed that it'll stay this way. Dunno what was the problem in version 3.x, but it's not present on 4.x

opdenkamp, 2016-11-07:
thanks for reporting back

closed 2016-11-07 by opdenkamp