Pulse-Eight/libcec #264: CEC-lost on Samsung JS9090

Kampfader, 2016-11-01:
hello,
first is, im not so good in englisch, sorry.
but i have a bug with CEC+Kodi17(Millhouse) #1031 on my TV.
--------
im usind Raspberry Pi3 connected --> HMDI1 on TV
i have a Samsung Soundbar connected ---> Bluetooth on TV
and a PS4 connected --> HDMI2 on TV
-------
i habe making a debug log with libCEC = on

I'm so doing this log..
TV OFF on HDMI1 --> Raspberry reboot -->TV ON (no switching the HMDI channels) = no CEC connected
i go on TV to sources HMDI1 put down and push on Anynet+ CEC --> choose kodi and i have CEC on Kodi, push 5 times on kodi menu down and 5 times up.
I go the second time to Sources on TV and choose the TV Sources for my DVB-C, after 5-10sec im going back to HDMI1 = no CEC connected on Kodi, I must again Anynet+ CEC on HMDI1 to connected this, I hope you understand me guys.

my log: http://pastebin.com/v4KUVtZL

opdenkamp, 2016-11-01:
I can see that you're switching to the TV's input at 22:41:47 and switch back to the Pi at 22:42:01. Then at 22:42:07 I can see the first key being pressed. That's 20 seconds after your switch to the TV, so I'm not sure what's not working exactly?

Kampfader, 2016-11-02:
im making a video, im not so good in to explain englisch.
i think you can see my problem.
today i updating the millhouse build to #1101, 
is the same problem.

log2: http://pastebin.com/2MU5UCjs
video: https://www.youtube.com/watch?v=NZn-zMRWgYc

opdenkamp, 2016-11-08:
This is a bug in your TV unfortunately. It's still sending keys to Kodi after you've selected the TV's internal source.

What happens when you select that Playstation first, then select the TV's internal source and then press the number buttons on your remote?

Kampfader, 2016-11-08:
When I first go to PlayStation and then on TV is OK, All commands are sent to TV and not to Pi or Ps4.
But when I go back to Kodi, the  connectionof Anynet+ CEC does not work.

I've also researched and it's likely Samsung will fail with the last Firmware ;(
thanks for helping @opdenkamp

opdenkamp, 2016-11-08:
Can you upload a log please in which you do this:
* select the Playstation
* press some buttons on the remote, like up/down/left/right
* select the TV
* press some buttons again
* select the Pi
* press some buttons
* back to the TV
* press some buttons
* back to the Pi

thanks

Kampfader, 2016-11-08:
ok, here is the new log, 
1. im connecting to Pi (cec), and reboot the Pi, after rebooting i have cec function.
2. processes the template = no Cec on Pi
3. connecting to cec manually on Pi, and press up/down/left/right
finish

log3: http://pastebin.com/tuD0U2tZ

opdenkamp, 2016-11-08:
I can see that the Pi was started correctly around 10:32:53, but you've got the option to make Kodi the active source disabled in Kodi's settings. This is preventing Kodi from taking active source status correctly, and the TV may send keys to another device here, or prevent Kodi from working at all until you select it in the TV's input menu.

This is expected behaviour.

Then I can see the first key from the TV coming in at 10:37:06. The TV's input is selected at 10:37:15. Then the TV tells the Pi become active again at 10:38:25. Then the first key is received by Kodi at 10:38:30.

I'm not seeing any issues in this log, and I don't see the TV selecting the Playstation or send any keys to it, even though the Playstation is responding to CEC queries from the TV correctly.

Did you select the Playstation in this last test? Where did the Pi fail to work in this last test?

Kampfader, 2016-11-08:
a ok, I misunderstood.
I need to connect to the PS4 with Anynet + (CEC)?
and now go to TV(up/down/left/right)-->Pi(up/down/left/right)-->TV(up/down/left/right)-->Pi(up/down/left/right)?

on the last test im connecting over CEC to Pi and go to PS4-->TV-->Pi-->TV-->Pi

opdenkamp, 2016-11-08:
The actions described in https://github.com/Pulse-Eight/libcec/issues/264#issuecomment-259085000 in sequence please. Then upload a new load and tell me which of the items don't work as expected (what's happening, what's you're expecting to happen)

Kampfader, 2016-11-08:
ok here a new log: http://pastebin.com/4taKqytt
im connected over cec to ps4 and press the up/down/left/right
change the source from ps4 to tv press the some buttons
change to Pi press the some buttons
change from Pi to TV press the some buttons
and the last go back to Pi and press the some buttons
and i have no CEC connection on Kodi

im realy think that is a tv firmware bug

jester-xbmc, 2016-12-11:
@opdenkamp Hi Lars, it might be the lastest samsung firmware on J series, look here:
https://forum.samygo.tv/viewtopic.php?f=85&t=11166

I'm running into a similar issue (intel nuc/ internal cec adapter/libreelec) having to manually select hdmi-cec on the input source for it to work (and even switch on the nuc)

glyph-se, 2016-12-15:
I have the same issue on my J6300 since the new firmware.

I found a thread on the LibreELEC forum that has a dirty patch to Kodi https://forum.libreelec.tv/thread-2387-post-19195.html#pid19195 . But it appears that the fix should be made in libCEC instead.

glyph-se, 2016-12-20:
My issues was solved by updating my J6300 to firmware 1480, available on Samsung's web page.

Kampfader, 2016-12-21:
I can confirm, tv Firmware 1480, solved my Problem. Js9090.

jester-xbmc, 2016-12-21:
Nope, issue not fixed for intel nuc with internal p8 cec board, still have to manually select cec using source button.

opdenkamp, 2017-01-07:
Can you check again with Kodi's latest nightly build please, using libCEC 4.0.1

Kampfader, 2017-01-11:
Do u mean me?
With me was tv firmware problem.
Im testing the last night with libcec 4.0.1
And its working perfekt, i have raspberry3 and Samsung js9090

opdenkamp, 2017-01-11:
sorry, I meant @jester-xbmc 
good to know that it's all working for you :)

jester-xbmc, 2017-01-15:
waiting for me new nuc, but looks ok with 4.01 and new firmware.
Thanks Lars.

opdenkamp, 2017-01-23:
thanks for reporting back

closed 2017-01-23 by opdenkamp