Pulse-Eight/libcec #266: Commit 2125cd requires -std=c++11

UEckhardt, 2016-11-10:
Hi,

with commit [https://github.com/Pulse-Eight/libcec/commit/736dd56be9c23ad5e471f86d83a3d2b2ae2125cd] the option -std=c++11 is required, since the change from NULL to nullptr requires C++11. Is this intended since libcec 3 compiled fine without c++11.

opdenkamp, 2016-11-10:
This changed 1.5yrs ago, in https://github.com/Pulse-Eight/libcec/commit/0b1be027deb8fc0f9120e6df884788fdc1ffc019 which requires c++11

The flag is added by https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/CMakeLists.txt#L16

closed 2016-11-10 by opdenkamp

UEckhardt, 2016-11-10:
I was unprecice with my description. It is not a problem of compiling libcec. Before this commit I could compile my application against libcec without the use of -std=c++11. After this commit I got the following errors and have to use -std=c++11 as compile option for my application.
```
In file included from cecremote.h:19:0,
                 from cecremote.cc:12:
/usr/include/libcec/cectypes.h:1424:5: warning: identifier 'nullptr' is a keyword in C++11 [-Wc++0x-compat]
     logMessage           = nullptr;
     ^
/usr/include/libcec/cectypes.h: In member function 'void CEC::ICECCallbacks::Clear()':
/usr/include/libcec/cectypes.h:1424:28: error: 'nullptr' was not declared in this scope
     logMessage           = nullptr;
                            ^
/usr/include/libcec/cectypes.h: In member function 'void CEC::libcec_configuration::Clear()':
/usr/include/libcec/cectypes.h:1553:21: error: 'nullptr' was not declared in this scope
     callbackParam = nullptr;
```

opdenkamp, 2016-11-10:
C++11 is required for compiling libCEC. I could change the headers so you don't need C++11 for your application, but only when compiling libCEC itself I suppose, but is there any good reason?

UEckhardt, 2016-11-11:
Since libCEC is a prerequisite for compiling my vdr plugin a C++11 capable compiler must already be present. But how about adding the -std=c++11 to the pkg-config for libCEC?

opdenkamp, 2017-07-13:
Very late response, but added now: https://github.com/Pulse-Eight/libcec/commit/456c7a62ad82f43f630a3e1224afb47f6d99082c

dougnazar, 2018-03-09:
This breaks my build, where I'm using -std=gnu++17. No other pkg-config file on my systems specifies a -std option.