Pulse-Eight/libcec #271: Discrepancy between the 'scan' and 'list all devices'

gordonturner, 2016-11-26:
First, thanks for a great tool, cec-client is very useful, I can throw away my smart remote.  Almost.

It appears that there is a discrepancy between the 'scan' and 'list all devices'.

## Background

I have a Raspberry Pi running Kodi and an older Tivo (that doesn't understand cec) with a Pulse Eight USB adapter.

The Pulse Eight USB adapter is connected to the Raspberry Pi running Kodi.

So the result is I have 1 computer with two cec devices. (Perhaps not typical?)

## Issue

I am trying to figure out transmitting frames (using the cec-o-matic) and obviously I need to use the correct addresses.  

It appears that the 'scan' command was incorrect and not providing complete information (missing the RPI) when using the Pulse Eight device.

## Analysis

When I started poking around at what address things had, I was getting mixed results, which confused me.

If `echo scan | cec-client /dev/ttyACM0 -s -d 1` is run, the Pulse Eight doesn't report the RPI and reports Pulse Eight address 1.0.0.0.

If `echo "scan" | cec-client RPI -s -d 1` is run, the RPI reports everything, but it reports RPI address 1.0.0.0 and Pulse Eight address 2.0.0.0.

So I started looking at the 'list all devices' command and what other single commands debug details say.

If `cec-client /dev/ttyACM0 -l -d 1` or `cec-client RPI -l -d 1` are run, they agree device 1 is Pulse Eight and device 2 is RPI

If `echo "on 0" | cec-client /dev/ttyACM0 -s -d 4` is run, logical address 1 and physical address 1.0.0.0 is reported for the Pulse Eight.

If `echo "on 0" | cec-client RPI -s -d 4` is run, logical address 2 and physical address 2.0.0.0 is reported for the RPI.

Using  libcec-4.0.0-3.tar.gz and p8-platform-2.1.0.1.tar.gz, compiled on a clean install of Rasbpain (2016-05-27-raspbian-jessie.img) with full updates applied.

```
cec-client -i
libCEC version: 4.0.0, git revision: <unknown>, compiled on Sat 26 Nov 20:02:07 UTC 2016 by pi@raspberrypi on Linux 4.4.32-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
```

## Notes

- Scan cec bus using the Pulse-Eight:

```
echo scan | cec-client /dev/ttyACM0 -s -d 1
```

- Result:

```
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      eng

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

device #4: Playback 1
address:       3.0.0.0
active source: no
vendor:        Sony
osd string:    PlayStation 4
CEC version:   1.3a
power status:  standby
language:      ???

currently active source: unknown (-1)
```

- Scan cec bus using the Raspberry Pi:

```
echo "scan" | cec-client RPI -s -d 1
```

- Result:

```
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      eng

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Unknown
osd string:    CECTester
CEC version:   1.4
power status:  standby
language:      ???

device #2: Recorder 2
address:       2.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

device #4: Playback 1
address:       3.0.0.0
active source: no
vendor:        Sony
osd string:    PlayStation 4
CEC version:   1.3a
power status:  standby
language:      ???

currently active source: unknown (-1)
```

List All Devices On This System
-------------------------------

-  List all devices on this system, using the Pulse-Eight:

```
cec-client /dev/ttyACM0 -l -d 1
```

```
libCEC version: 4.0.0, git revision: <unknown>, compiled on Sat 26 Nov 20:02:07 UTC 2016 by pi@raspberrypi on Linux 4.4.32-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
Found devices: 2

device:              1
com port:            /dev/ttyACM0
vendor id:           2548
product id:          1002
firmware version:    5
firmware build date: Tue Jan 27 16:14:39 2015 +0000
type:                Pulse-Eight USB-CEC Adapter

device:              2
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi

```

-  List all devices on this system, using the RPi:

```
cec-client RPI -l -d 1
```

```
libCEC version: 4.0.0, git revision: <unknown>, compiled on Sat 26 Nov 20:02:07 UTC 2016 by pi@raspberrypi on Linux 4.4.32-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
Found devices: 2

device:              1
com port:            /dev/ttyACM0
vendor id:           2548
product id:          1002
firmware version:    5
firmware build date: Tue Jan 27 16:14:39 2015 +0000
type:                Pulse-Eight USB-CEC Adapter

device:              2
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi
```

Single Command Device Details `/dev/ttyACM0`
--------------------------------------------

- Device details based on calls to `/dev/ttyACM0`:

```
echo "on 0" | cec-client /dev/ttyACM0 -s -d 4
```

- Parsed:

```
logical address(es) = Recorder 1 (1) 
base device: TV (0)
HDMI port number: 1
physical address: 1.0.0.0
```

Single Command Device Details `RPI`
-----------------------------------

- Device details based on calls to `RPI`:

```
echo "on 0" | cec-client RPI -s -d 4
```

- Parsed:

```
logical address(es) = Recorder 2 (2)
base device: N/A
HDMI port number: N/A
physical address: 2.0.0.0
```

opdenkamp, 2017-01-02:
Could you build libCEC from the master branch please and, if the issue still occurs, upload the output with any `-d` options please.

One thing to keep in mind is that the USB-CEC adapter can only read the EDID of the TV if randr is running, so if the gui is running and you run cec-client with from the correct X display. libCEC will use address 1.0.0.0 for the adapter if it can't read the EDID. That may cause a collision with another device.

You have to use the `-p` and possibly the `-b` parameter if cec-client isn't started from X when using the USB-CEC adapter. The Pi's internal wire doesn't require this.

gordonturnerolson, 2017-01-04:
Will do, should have time to do this tomorrow and report back.

gordonturnerolson, 2017-01-06:
Did a build based on:

https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

Looks good to me!   The scan results from `echo scan | cec-client /dev/ttyACM0 -s -d 1` and `echo "scan" | cec-client RPI -s -d 1` are now consistent.

- Scan cec bus using the Pulse-Eight:

```
echo scan | cec-client /dev/ttyACM0 -s -d 1
```

```
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      eng

device #1: Recorder 1
address:       2.0.0.0
active source: yes
vendor:        Unknown
osd string:    raspberry
CEC version:   1.3a
power status:  on
language:      eng

device #2: Recorder 2
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

device #4: Playback 1
address:       3.0.0.0
active source: no
vendor:        Sony
osd string:    PlayStation 4
CEC version:   1.3a
power status:  standby
language:      ???

currently active source: Recorder 1 (1)
```

- Scan cec bus using the Raspberry Pi:

```
echo "scan" | cec-client RPI -s -d 1
```

```
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      eng

device #1: Recorder 1
address:       2.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

device #2: Recorder 2
address:       1.0.0.0
active source: no
vendor:        Unknown
osd string:    CECTester
CEC version:   1.4
power status:  standby
language:      ???

device #4: Playback 1
address:       3.0.0.0
active source: no
vendor:        Sony
osd string:    PlayStation 4
CEC version:   1.3a
power status:  standby
language:      ???

currently active source: unknown (-1)
```

-  List all devices on this system, using the Pulse-Eight:

```
cec-client /dev/ttyACM0 -l -d 1
```

```
libCEC version: 4.0.1, git revision: libcec-4.0.1, compiled on Thu  5 Jan 23:41:56 UTC 2017 by pi@raspberrypi on Linux 4.4.38-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
Found devices: 2

device:              1
com port:            /dev/ttyACM0
vendor id:           2548
product id:          1002
firmware version:    5
firmware build date: Tue Jan 27 16:14:39 2015 +0000
type:                Pulse-Eight USB-CEC Adapter

device:              2
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi
```

-  List all devices on this system, using the RPi:

```
cec-client RPI -l -d 1
```

```
libCEC version: 4.0.1, git revision: libcec-4.0.1, compiled on Thu  5 Jan 23:41:56 UTC 2017 by pi@raspberrypi on Linux 4.4.38-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
Found devices: 2

device:              1
com port:            /dev/ttyACM0
vendor id:           2548
product id:          1002
firmware version:    5
firmware build date: Tue Jan 27 16:14:39 2015 +0000
type:                Pulse-Eight USB-CEC Adapter

device:              2
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi
```

opdenkamp, 2017-01-06:
thanks for reporting back and good to hear that it's been fixed

closed 2017-01-06 by opdenkamp