Pulse-Eight/libcec #273: Not working on RasPi 3

spacelego, 2016-12-04:
Fresh install of 2016-11-25-raspbian-jessie-lite

sudo apt-get install cec-utils

...went ok, seems like I got v4.0, anything I try to run I get...

autodetect FAILED

TV shows "raspberry"
Tried different cables

Not sure what to do.

popcornmix, 2016-12-04:
I don't believe that version of libcec comes from either upstream raspbian, or the raspberry pi repo.
I suspect you have added an additional repo that contained this (perhaps pipplware or Michael Gorven's kodi repo?)
The issue probably needs to be reported with them first, in case they have made a mistake in configuring libcec.

opdenkamp, 2017-01-02:
Try again with libCEC built from current master. If this still occurs, please send me the output of `cec-client -d 31 -l`

closed 2017-01-02 by opdenkamp