Pulse-Eight/libcec #276: pulse eight usb adapter not working since libcec 4.0

sopparus, 2016-12-07:
Hello, just upgraded from libreelec alpha 008 to 009 and the cec adapter stopped working (x86 generic nuc)

Here is log from libcec 4.0 (le 009) -> http://sprunge.us/NLNJ

and here is with libcec 3.1 (le 008) - > http://sprunge.us/ONDU

I did notice this with libcec 4.0 log
GetPhysicalAddress - OS returned physical address 0000

doesnt look right?

thanks for any help

opdenkamp, 2016-12-11:
Found a problem in the physical address detection code that's probably causing this. Will fix it.

yasij, 2016-12-19:
Could this also by why Kodi keeps resetting the physical address for my CEC adapter to 1000 on exit or suspend/resume cycle despite manually setting it to 2500 in settings?
Kodi bug: http://trac.kodi.tv/ticket/17144

sakos, 2016-12-27:
Same problem on my system. TV swicthes to wrong HMDI input. CEC works if I set the HDMI port manually in CEC settings. Please correct and make it work automatically as it worked before.

Meindert66, 2017-01-01:
I also have problems with the new libcec in combination with kodi 17.  Kodi 16 with libcec 3.1 works, most of the time, good. Sony Bravia TV shows two sync divices, both on HDMI4 with libcec3, but with libcec4 one device is on HDMI1 and the other on HDMI4. The DENON AVR is connected to HDMI4 because of reverse audio.
I have no idea what number to enter in physical HW adress! But there are many options without any explanation and that makes configuration difficult.

opdenkamp, 2017-01-02:
Should be fixed now with current master, but I'll check with Kodi tonight.

opdenkamp, 2017-01-03:
Looks like it'll have to wait until tomorrow. Kodi's cmake changes are broken, finding a library, building support for it but not linking it...

Meindert66, 2017-01-03:
Thanks. After compiling kodi and libcec from todays source it seems to work as with previous version.

opdenkamp, 2017-01-03:
Thanks for verifying. I'll close this ticket then and issue a 4.0.1 update later today

closed 2017-01-03 by opdenkamp

opdenkamp, 2017-01-03:
PR is up here: https://github.com/xbmc/xbmc/pull/11361

Portisch, 2017-01-09:
Problem isn't fixed on Windows. Please take a look to the Kodi bugtracker.