Pulse-Eight/libcec #277: Remote buttons not working on TCL TV

jbmfg, 2016-12-07:
Im betting this is an issue with my TV rather than libcec.  

Command echo "standby 0" | cec-client -s -p 2 does work to put my tv into standby so i know cec is working.  However no remote button presses are registered.  Not sure what to do to troubleshoot

greedysquid, 2016-12-09:
I believe I have the same problem/symptom.

Panasonic TX-P42G20B TV
Raspberry Pi B+ v1.2
Clean install of 2016-11-25-raspbian-jessie-lite.zip
Then install CEC utilities: `sudo apt-get cec-utils`

My goal is to use the above RaspberryPi to initially monitor and then interact with CEC traffic.
CEC is already working on my setup; the TV can control an amp, DVD player and another Kodi RaspberryPi.

Running `cec-client` successfully initialises and auto-configures itself on to the CEC bus.

`pi@raspberrypi:~ $ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 3.0.1
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [             135]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             140]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [             141]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [             142]     InitHostCEC - vchi_connect succeeded
DEBUG:   [             145]     logical address changed to Free use (e)
DEBUG:   [             147]     Open - vc_cec initialised
NOTICE:  [             149]     connection opened
DEBUG:   [             151]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [             151]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             151]     << e0
DEBUG:   [             156]     processor thread started
DEBUG:   [             182]     >> POLL sent
DEBUG:   [             182]     TV (0): device status changed into 'present'
DEBUG:   [             182]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [             182]     << e0:8c
TRAFFIC: [             391]     >> 0f:87:00:80:45
DEBUG:   [             391]     TV (0): vendor = Panasonic (008045)
DEBUG:   [             392]     expected response received (87: device vendor id)
DEBUG:   [             392]     replacing the command handler for device 'TV' (0)
NOTICE:  [             392]     registering new CEC client - v3.0.1
DEBUG:   [             392]     detecting logical address for type 'playback device'`

`NOTICE:  [            1572]     CEC client registered: libCEC version = 3.0.1, client version = 3.0.1, firmware version = 1, logical address(es) = Playback 3 (B) , physical address: 4.0.0.0,  compiled on Fri Jul  3 19:20:43 UTC 2015 by pbuilder@mgorven-pc.thefacebook.com on Linux 3.16.0-41-generic (armv7l), features: P8_USB, P8_detect, 'RPi'`

The above shows that the Pi is connected-to, and sending/receiving-on, the CEC bus. Manually sending a command (eg wakup DVD player) is also sucessful.

However, when pressing a button on the TV's remote nothing is seen in the output of cec-client.
A command is definately sent over the CEC bus - eg volup makes the amplifier change volume, and using a scope I can see corresponding traffic on the CEC pin of the Pi.

So, like @jbmfg who opened this issue (#277), I too do not see remote control button press events in the output from cec-client (from the current Raspian cec-utils package).

How to debug further?

opdenkamp, 2017-01-02:
cec-client will not automatically make itself the active source on the CEC bus when started, so it won't receive any keys from the TV by default in most situations.

You have to tell cec-client to become active (the command `as`), then it'll work.

closed 2017-01-02 by opdenkamp