Pulse-Eight/libcec #280: MegaChips MDCP2800 LSPCON  driver

9000h, 2016-12-21:
is there any plan to support the MegaChips MDCP2800 LSPCON chip?

the new NUC6CAYH will also have it and CEC support should be possible as there is support in the Bios
https://www-ssl.intel.com/content/dam/support/us/en/documents/boardsandkits/NUC6CAYB_TechProdSpec.pdf

there are also some motherboards from Gigabyte and Asrock with this DP>HDMI chip out

opdenkamp, 2017-01-02:
as far as I can tell from the docs, the pin is simply exposed by the chip and doesn't contain any logic, so it would have to be bitbanged by some kernel implementation and then be exposed to libCEC.
the new Linux in-kernel CEC implementation would be the right place to do this.

closed 2017-01-02 by opdenkamp

9000h, 2017-06-24:
there is a driver coming to the kernel soon
https://patchwork.freedesktop.org/project/intel-gfx/patches/?submitter=15857

opdenkamp, 2017-07-13:
I'm sure that it'll use the new in-kernel framework then, for which there is a ticket here: https://github.com/Pulse-Eight/libcec/issues/67