Pulse-Eight/libcec #283: libCEC instability problems on RPI

moosebert, 2016-12-29:
Fresh RPI 2 install and libCEC build.  Noticing very frequent failures due to SIGABRT returns, segfaults, etc.

libCEC version: 4.0.0, git revision: libcec-4.0.0-3+4-12fa275, compiled on Thu 29 Dec 01:35:09 UTC 2016 by pi@retropie on Linux 4.4.26-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi

Attaching output examples.
[libcec-output.txt](https://github.com/Pulse-Eight/libcec/files/677970/libcec-output.txt)

Followed https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md to the letter for the build.

opdenkamp, 2017-01-02:
Could you try again with libCEC's current master please.

If it still happens, please run cec-client in gdb, and then send me the stack trace: `thread apply all bt`

moosebert, 2017-01-02:
Snagged a copy as requested and built fresh.

libCEC version: 4.0.0, git revision: libcec-4.0.0-3+5-ab18a95, compiled on Mon Jan  2 22:53:56 UTC 2017 by pi@retropie on Linux 4.4.26-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi

Unfortunately still experiencing stability issues.  Kindly see attached output.

[output.txt](https://github.com/Pulse-Eight/libcec/files/681469/output.txt)

Will go after the debug requested in a bit.

opdenkamp, 2017-01-02:
The Pi's firmware is responding with "busy" at first. Do you perhaps have something else using CEC, like Kodi? If so, then that won't work and we need to fix the check to tell this and exit.

moosebert, 2017-01-02:
Funny thing, the scan will run sometimes albeit ending with abort message (same as the output in my first post.  When running libCEC 3.0.1 (current Raspberry repo package release) I experienced segfaults due to known DRM class directory missing issues, but otherwise it was stable.  Nothing like what I'm seeing now.

Regarding your question regarding something else using CEC, nope.  The AppleTV is the only other CEC player in the equitation.

opdenkamp, 2017-01-02:
Right ok, can you try again without that cron job running. I suspect that it was causing the "busy" problem here. Then upload a new log. Thanks

moosebert, 2017-01-02:
Absolutely.  Confirmed no conflicting CEC communication at the time.

pi@retropie:~ $ echo 'scan' | cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              78]	Broadcast (F): osd name set to 'Broadcast'
ERROR:   [              79]	RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
ERROR:   [              79]	Open - vc_cec could not be initialised
ERROR:   [              79]	could not open a connection (try 1)
... (ten tries)
unable to open the device on port RPI
NOTICE:  [           10086]	connection opened
ERROR:   [           10086]	could not start CEC communications

opdenkamp, 2017-01-03:
Not just communication, but is anything else running that's using libCEC or calling any of the `vc_set_*()` methods?
I don't see how a change in libCEC could be causing this. Not seen any other reports, so I'm pretty sure that it's something local to your setup.

opdenkamp, 2017-01-03:
Looks like I forgot about this one https://github.com/Pulse-Eight/libcec/issues/191 and it was happening with v3 too. Closing this ticket as dupe.
I couldn't reproduce it on a vanilla Raspbian Jessie installation when I checked this, so I'm assuming that it's some change in your distribution that causes trouble.

closed 2017-01-03 by opdenkamp