Pulse-Eight/libcec #288: Python cec module is installed into a Debian specific path

armills, 2017-01-09:
The python module install path is hardcoded to go into a directory named `dist-packages`, which is only used by Debian and derivatives.

[There are some examples here](http://stackoverflow.com/questions/1242904/finding-python-site-packages-directory-with-cmake) of how to use python within cmake to fetch the correct packages directory from within cmake.

Unless there are any objections I'm going to put together a PR to load the correct directory dynamically from the current python executable.

See also:
https://bugs.gentoo.org/show_bug.cgi?id=577612

opdenkamp, 2017-01-23:
this has been corrected in the master branch, thanks

closed 2017-01-23 by opdenkamp

armills, 2017-01-23:
Cool. Thank you!