Pulse-Eight/libcec #290: Windows, Kodi Krypton does not find the adapter

Portisch, 2017-01-11:
The adapter doesn't get loaded on windows systems:

> 20:20:05.086 T:5488   DEBUG: PERIPHERALS::CPeripherals::LoadMappings - loaded node "Motorola Nyxboard Hybrid"
> 20:20:05.086 T:5488   DEBUG: PERIPHERALS::CPeripherals::LoadMappings - loaded node "CEC Adapter"
> 20:20:05.087 T:5488   DEBUG: PERIPHERALS::CPeripherals::LoadMappings - loaded node "Pulse-Eight CEC Adapter"
> 20:20:05.087 T:5488   DEBUG: PERIPHERALS::CPeripherals::LoadMappings - loaded node "iMON HID device"
> 20:20:05.088 T:5488   DEBUG: PERIPHERALS::CPeripherals::LoadMappings - loaded node "WETEK Play remote"
> 20:20:05.088 T:5488   DEBUG: PERIPHERALS::CPeripherals::LoadMappings - loaded node ""
> 20:20:05.088 T:5488   DEBUG: SECTION:LoadDLL(special://xbmcbin/libcec.dll)
> 20:20:05.099 T:5488   DEBUG: Add-on bus: Registering add-on peripheral.joystick
> 20:20:05.100 T:5488   DEBUG: PERIPHERAL - PERIPHERALS::CPeripheralAddon::CreateAddon - creating peripheral add-on instance 'Joystick Support'
> 20:20:05.100 T:5488   DEBUG: ADDON: Dll Initializing - Joystick Support
> 20:20:05.101 T:5488   DEBUG: SECTION:LoadDLL(C:\Program Files (x86)\Kodi\addons\peripheral.joystick\peripheral.joystick.dll)
> 20:20:05.106 T:5488   DEBUG: AddOnLog: Joystick Support: Attempting to load XInput1_4.dll...
> 20:20:05.107 T:5488   DEBUG: AddOnLog: Joystick Support: Attempting to load XInput1_3.dll...
> 20:20:05.109 T:5488   DEBUG: AddOnLog: Joystick Support: Loaded XInput DLL version 1.3
> 20:20:05.109 T:5488   DEBUG: AddOnLog: Joystick Support: Loaded XInput symbols (GetState=04BE2AA0, SetState=04BE2C20, GetCaps=04BE2DA0, GetBatteryInformation=04BE3160, PowerOff=04BE3040)
> 20:20:05.111 T:5488   DEBUG: AddOnLog: Joystick Support: Loaded 14 joystick families with 97 total joysticks
> 20:20:05.111 T:5488   DEBUG: Calling TransferSettings for: Joystick Support
> 20:20:05.115 T:5488   DEBUG: SECTION:LoadDLL(special://xbmcbin/libcurl.dll)
> 20:20:05.116 T:3740   DEBUG: Thread PeripBusUSB start, auto delete: false
> 20:20:05.117 T:3856   DEBUG: Thread PeripBusAddon start, auto delete: false
> 20:20:17.385 T:6192   DEBUG: CRemoteControl::Connect - failed to connect
> 20:20:18.292 T:6192    INFO: CRemoteControl::Process - failed to connect to irss, will keep retrying every 5 seconds
> 20:20:33.381 T:3740   DEBUG: PERIPHERALS::CPeripherals::GetMappingForDevice - device (2548:1002) mapped to Pulse-Eight CEC Adapter (type = cec)

Nothing else is happening!

In Kodi source "Peripherals.cpp", function "void CPeripherals::CreatePeripheral(CPeripheralBus &bus, const PeripheralScanResult& result)" the device is coming in:

> -		mappedResult	{m_type=PERIPHERAL_UNKNOWN (0x00000000) m_strLocation="\\\\?\\usb#vid_2548&pid_1002#v2_r8#{a5dcbf10-6530-11d2-901f-00c04fb951ed}" ...}	PERIPHERALS::PeripheralScanResult
> 		m_type	PERIPHERAL_UNKNOWN (0x00000000)	PERIPHERALS::PeripheralType
> +		m_strLocation	"\\\\?\\usb#vid_2548&pid_1002#v2_r8#{a5dcbf10-6530-11d2-901f-00c04fb951ed}"	std::basic_string<char,std::char_traits<char>,std::allocator<char> >
> 		m_iVendorId	0x00002548	int
> 		m_iProductId	0x00001002	int
> 		m_mappedType	PERIPHERAL_CEC (0x00000005)	PERIPHERALS::PeripheralType
> +		m_strDeviceName	"Pulse-Eight CEC Adapter"	std::basic_string<char,std::char_traits<char>,std::allocator<char> >
> 		m_busType	PERIPHERAL_BUS_USB (0x00000001)	PERIPHERALS::PeripheralBusType
> 		m_mappedBusType	PERIPHERAL_BUS_USB (0x00000001)	PERIPHERALS::PeripheralBusType
> 		m_iSequence	0x00000000	unsigned int
> 		(bus).m_type	PERIPHERAL_BUS_USB (0x00000001)	const PERIPHERALS::PeripheralBusType

But the bus:

> -		bus	{...}	PERIPHERALS::CPeripheralBus & {PERIPHERALS::CPeripheralBusUSB}
> +		[PERIPHERALS::CPeripheralBusUSB]	{...}	PERIPHERALS::CPeripheralBusUSB
> +		CThread	{m_bStop=false m_ThreadId=0x00000840 m_ThreadOpaque={handle=0x00000854 } ...}	CThread
> +		m_peripherals	{ size=0x00000000 }	std::vector<std::shared_ptr<PERIPHERALS::CPeripheral>,std::allocator<std::shared_ptr<PERIPHERALS::CPeripheral> > >
> 		m_iRescanTime	0x00001388	int
> 		m_bInitialised	false	bool
> 		m_bIsStarted	true	bool
> 		m_bNeedsPolling	true	bool
> +		m_manager	0x01a5c288 {kodi.exe!PERIPHERALS::CPeripherals peripheralsInstance} {m_bInitialised=true m_bIsStarted=...}	PERIPHERALS::CPeripherals * const
> 		m_type	PERIPHERAL_BUS_USB (0x00000001)	const PERIPHERALS::PeripheralBusType
> +		m_critSection	{...}	CCriticalSection
> +		m_triggerEvent	{manualReset=true signaled=false numWaits=0x00000000 ...}	CEvent

bus.Type() == PERIPHERAL_BUS_USB  != PERIPHERAL_BUS_CEC
and the adapter never get initialised

Portisch, 2017-01-11:
Additional infos:
I found the bug!

The function
`uint8_t CUSBCECAdapterDetection::FindAdaptersWindows(cec_adapter_descriptor* deviceList, uint8_t iBufSize, const char* strDevicePath /* = NULL */)`
is stepping through all devices on the PC.
And the VID and PID compare is done in this function:
`static bool GetPidVidFromDeviceName(const std::string strDevName, int* vid, int* pid)`

But the device shows up this way:
> usb\vid_2548&pid_1002&mi_00\6&249ce8ee&0&0000

And the compare fails because of upper/lower case compare!

malard, 2017-01-11:
I think thats some really old code i wrote many years ago, my bad!

opdenkamp, 2017-01-23:
this has been fixed in master and will be included in the 4.0.2 update

closed 2017-01-23 by opdenkamp