Pulse-Eight/libcec #292: LG: cant switch the source on my TV.

LipkeGu, 2017-01-21:
When i have the RaspberryPI (with cec enabled) as active source and i want to change to TV input source (example XBOX360) the TV is switching back to the Raspberry PI.

The issue is happening always when the RaspberryPI is the active source.
The Dialog i get is "Connecting to Simplink device".

TectonicTMI, 2017-04-03:
I'm having the same issues on HASSBIAN (Rasbian Jessie-based Home Assistant distro), latest versions of everything. I can turn the TV on (off isn't supported on TVs, for whatever reason...), but once on, I can't change the source at all, it just stays on the Pi. 

I haven't really started debugging and reading logs, but if I find something, I will update you.

kkkkamal, 2017-06-03:
I need to enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.

opdenkamp, 2017-07-12:
It's the same as this one: https://github.com/Pulse-Eight/libcec/issues/307
This report is older, but the other one contains more info, so I'm closing this one.

closed 2017-07-12 by opdenkamp