Pulse-Eight/libcec #294: Multimeda Keys of Samsung TV not working

ircpsycho, 2017-01-21:
Hi, i running libcec 4.0 and the play, pause and stop button not working.
In Kodi 17 worked, but i exit and open libCEC Tray not.

Here the output of the stop button:

DEBUG - Main::onCecLogMessage(72315 [T]>> 01:42:03)
DEBUG - Main::onCecLogMessage(72316 [D]>> TV (0) -> Recorder 1 (1): deck control (42))
DEBUG - Main::onCecCommand(Command 0->1[ ] 66)

play button:

DEBUG - Main::onCecLogMessage(73585 [T]>> 01:41:24)
DEBUG - Main::onCecLogMessage(73586 [D]>> TV (0) -> Recorder 1 (1): play (41))
DEBUG - Main::onCecCommand(Command 0->1[ ] 65)

and for example some other buttons

DEBUG - Main::onCecLogMessage(21081 [T]>> 01:44:00)
DEBUG - Main::onCecLogMessage(21081 [D]>> TV (0) -> Recorder 1 (1): user control pressed (44))
DEBUG - Main::onCecCommand(Command 0->1[ ] 68)
DEBUG - Main::onCecLogMessage(21082 [D]key pressed: select (0))
DEBUG - Main::onCecKeyPress(Key press: SELECT for 0ms)
DEBUG - send 352
DEBUG - Main::onCecLogMessage(21784 [T]>> 01:8b:00)
DEBUG - Main::onCecLogMessage(21787 [D]>> TV (0) -> Recorder 1 (1): vendor remote button up (8B))
DEBUG - Main::onCecCommand(Command 0->1[ ] 139)
DEBUG - Main::onCecLogMessage(21789 [D]key released: select (0))
DEBUG - Main::onCecKeyPress(Key press: SELECT for 707ms)
DEBUG - release 352

Where can i change something, so that this button will cause an keypress action?

Thanks, and sorry for my english

opdenkamp, 2017-07-12:
Just checked what needs to be done for this. These commands don't send key press CEC commands, but play commands, which are currently not hooked in LibCecSharp. It first has to be added to LibCecSharp in the next major release, and then cec-tray needs to be updated to include these new methods.

Unfortunately this will have to wait until the next major release.

ircpsycho, 2017-07-12:
Thanks for responds. 
I'll wait for the major release.

closed 2025-02-27 by github-actions[bot]