Pulse-Eight/libcec #296: CEC-Adapter crashing KODI after standby

malansk, 2017-01-30:
Using KODI 17 with Pulse-Eight USB CEC adapter, WakeUp of KODI via TV Remote will often lead to crash of KODI.
Unplugging CEC adapter from USB while KODI is running will crash KODI 100%. Tried on two different PC's.

Is there anything that can be done from libCEC (4.0.0 and 4.0.1) to solve these issues ?

[http://trac.kodi.tv/ticket/17274](url)

[http://trac.kodi.tv/ticket/17278](url)

[http://forum.kodi.tv/showthread.php?tid=305205](url)

opdenkamp, 2017-01-30:
according to the tickets, it worked in rc3 and stopped working in rc4. nothing has been changed in the CEC implementation between these versions. The last change in Kodi's CEC implementation was included in beta6 and libCEC itself got bumped to 4.0.1 in rc3. This suggests that it's caused by something else.

I'm going to create a new pull request for libCEC 4.0.2, but nothing changed between these versions that should be the cause of a crash. Equally nothing changed between 4.0.0 and 4.0.1 that explains this.

Kodi never supported the `OnSleep` callback correctly on Windows, and fired it after Windows resumed instead of before. If that's still the case, then it's possibly colliding with the code in libCEC that reconnects to the adapter after the connection was lost, which is the case when coming out of standby without closing the connection before entering standby.

Can you post a crashlog with CEC logging enabled in it, so I can see what's happening exactly.

malansk, 2017-01-30:
Thanks for taking your time to look into this issue, some days ago I also noticed that the OnSleep callback looked wrong (after resume instead of before).

I have uploaded additional logfiles with componentspecific logging enabled as requested.

[http://trac.kodi.tv/attachment/ticket/17278/](url)

I will upload some more logs later with different versions of KODI (beta6, 15.2 14.x) since somebody mentioned cec got broken since update from libCEC 2.x to 3.x and 4.x.

malansk, 2017-01-30:
Updated trac ticket with logs from KODI14 (libCEC 2.2.0) as this was reported to be the last stable working combination of KODI & CEC Adapter for standby/resume.

Looks like KODI14 Standby/resume is stable but no detection if CEC Adapter replugged.

KODI15/16 (libCEC 3.x) fixed the replugging detection but broke detection after resume.

KODI17 (libCEC 4.x) tried to fix detection after resume but sometimes conflicting during initialization since both KODI and libCEC try to reconnect to device and KODI handling of OnSleep etc. events different than libCEC would expect.

Please let me know if I can provide some more information to help to improve this topic.

opdenkamp, 2017-07-12:
Sorry for the late response. I've checked the logs, and this looks like it's caused by another device on the CEC bus. The adapter reports that the line is high, when it should be low. This can't be caused by libCEC, but it looks like it is crashing as a result of this (which seems to be a new bug, not happening in 2.2.0).

You may need a CEC-less HDMI cable between the adapter to PC to work around this problem. First try updating your display card drivers to the latest version and reboot. Then try again.

If that doesn't help, try disconnecting all HDMI devices, so only the TV, PC and CEC adapter are attached. Does that help?

malansk, 2017-09-16:
Sorry for late response, yes, there is another CEC device on the bus, Denon AVR.
But when I switch Software environment from Windows  to Linux Mint, everything is working for several days without any crashes. 

Final configuration is now using a Pi3 which is on 24/7.

mcintyre321, 2019-10-08:
I'm having this problem, Kodi stops responding on wake-from-CEC on Windows 10. Just the TV, adapter and pc connected.

gibxxi, 2020-10-18:
Also having this issue, with a difference. For me, the system doesn't even need to get as far as sleep / standby mode. It's enough for Kodi to have sent the message to the AVR & TV to power down. As soon as a remote control key (like the OK button) is pressed and the TV and AVR attempt to wake, Kodi crashes / freezes, and takes the rest of the PC (Windows 10) with it, requiring a Power Off / reset.

For clarity;
Screensaver: 15 minutes,
TV switch off: 30 Minutes,
Standby 60 minutes.

It's not limited to Kodi either. Similar behaviour is occuring under Windows 10, desktop conditions with the newer 6.x.x libCEC version and v12 firmware. But it was also broken under 4.0.3 and 4.0.7, so i'm beginning to think a recent Windows update / build update might be the root cause here.

I have tried all versions of the Windows libCEC drivers package to attempt to eliminate the issue, but nothing has solved it. I also tried disabling all power-saving and ASPM features on the motherboard and in Windows, as to begin with I figured it might be a power issue, but nothing works or shows any difference in outcome (consistently).

I'm convinced there is nothing wrong with the hardware my HTPC is running on, as the only instability is at times the CEC adapter is being called into play. Remove it and the system works fine, albeit with a blue screen on my TV, at power saving / display suspend times, which I don't want to see.

closed 2025-02-27 by github-actions[bot]