Pulse-Eight/libcec #298: EventGhost plugin must reinitialze libcec after wake

dansrogers, 2017-01-31:
EventGhost loses it's connection to the device after a sleep event.  This prevents the TV from switching inputs on wake, which is the most important feature for me.  I suspect the hardware state is lost after a sleep event and device power down.  Though it is possible that the ping loop just isn't waiting long enough.

Regardless having an action that reinitializes the library lets me recover from all kinds of hardware events, including devices getting unplugged, so it seems handy regardless.  I made this work in my fork.  I saved the parameters to __start__ as member variables, then created member function ReinitLibCec and the corresponding action that just called lib.Close() then called InitLibCec with the last parameters called to __start__.  I can submit a pull request for the ~10 lines or so, but I do not know if I'll be able to sign the contributor agreement as I'll have to go though my employer.   I will work on this, this week.

After the change I can create a macro that watches for a resume event, reinitializes libcec and sets the active view, thus switching my TV to the right input (yay!)

opdenkamp, 2017-01-31:
Looks useful, thanks. Since it's straight forward code that can't be done in any other way, I should be able to add this without copyright issues and require your employer to agree :)

PS: for the Python plugin, please ensure that you're on libCEC 4.0.2 (current master). The generated Python code didn't have threading support in the previous versions

dansrogers, 2017-01-31:
Added pr/299

dansrogers, 2017-01-31:
Where can I find a build for libCEC 4.0.2 for windows?  The latest on the website is 3.0.1.  I'm trying to figure out the windows build, but I am stuck on locating the prerequisites (not normally a windows developer)

opdenkamp, 2017-01-31:
thanks!
it's not been uploaded yet because of some issues on other platforms. this will be uploaded soon: https://dl.dropboxusercontent.com/u/59212404/libCEC-4.0.2.exe (after a rebuild to include the latest docs and such)

dansrogers, 2017-05-04:
I'm just curious if there is anything that needs to be done on my end to close this issue.

opdenkamp, 2017-07-12:
It's been included in the master branch, and it'll be included in the next libCEC version. Thanks

closed 2017-07-12 by opdenkamp