Pulse-Eight/libcec #300: How to install the python module?

andrew-bolin, 2017-02-01:
I'm using a Raspberry Pi, have successfully got cec-client running by following the Raspberry Pi install instructions.

However, I can't figure out how to get the python module.
Trying to run the included pyCecClient.py gives
"""
Traceback (most recent call last):
  File "./pyCecClient.py", line 35, in <module>
    import cec
ImportError: No module named cec
"""

Any advice?
I'm guessing there's some other "make" target I need to build?

andrew-bolin, 2017-02-02:
Some progress:
> ~/libcec/src/libcec $ swig -I../../include -python -c++ libcec.i

This builds a cec.py file. Can't use it though...

> Traceback (most recent call last):
  File "/home/pi/cec.py", line 18, in swig_import_helper
    fp, pathname, description = imp.find_module('_cec', [dirname(__file__)])
  File "/usr/lib/python3.4/imp.py", line 297, in find_module
    raise ImportError(_ERR_MSG.format(name), name=name)
ImportError: No module named '_cec'

opdenkamp, 2017-03-15:
Most likely issue is that your `PYTHONPATH` variable is wrong. Please use the binary that's available in the apt repository if you're having trouble building it manually.

closed 2017-03-15 by opdenkamp

andrew-bolin, 2017-03-15:
Is there a special libcec apt repository somewhere? (If so, where can I
find its details?)

I can't get the version of libcec included my distribution (Raspbian) to
work either, hence why I tried building a fresh one.
But I will look into the hack suggested in issue #316 to try and get the
import to work...