Pulse-Eight/libcec #304: CEC issue on UA55KS9000 Samsung TV

haioken, 2017-02-07:
I'm lodging this as an issue, as it seems to be directly related to LibCEC (Not Kodi), and I've had little success with requests to pulse-eight support.

I have recently upgraded my TV to a Samsung UA55KS9000.
With no devices attached aside from a Linux system running Pulse-Eight CEC controller firmware version 4 and LibCEC 4.0.2, the TV switches to the correct source, however only displays content briefly before saying 'No source'. After about 10 seconds, the TV will display content again for a fraction of a second before again saying 'No Source'.

Log file attached running CEC-Client from the command line.
The only potential issue I can see is that the CEC controller reports a version that cec-o-matic.com states is 'unexpected' (10:9e:05: CEC Adapter -> TV, Indicate CEC version 05).

[ceclog2.txt](https://github.com/Pulse-Eight/libcec/files/756314/ceclog2.txt)

opdenkamp, 2017-03-15:
Sounds like a problem with your HDMI cables. Try to swap both cables between the TV and adapter and the adapter and PC.

If you're now using 4K/HDR, then you'll need good quality cables rated to work with 4K video. The small HDMI cable that you've got with the adapter does not support 4K.

closed 2017-03-15 by opdenkamp

haioken, 2017-03-15:
Hi Lars,

Frankly, having asked questions in multiple places (Multiple emails to your
support department, this github bug report, and the Kodi forums) and not
receiving anything useful for *over a month*, I have given up on the idea.
The CEC controller has been disposed of, and my media center replaced with
an Nvidia shield.

It's worth noting that I had attempted using different HDMI cables on both
input and output sides of the CEC controller with no improvement.

Regards,

Hayden Katz