Pulse-Eight/libcec #306: No CEC support after turn on TV from standby

NewFolk, 2017-02-08:
Hello,
I updated my raspberry pi 2 to kodi krypton (osmc) with 4.0.2 LibCEC.

Now when i go to standby mode on TV (power button) and then power on TV I lost CEC support.
I could resume it only by select HDMI CEC input again.

1) turn on raspberry 
2) run "cec-client -r"
3) check CEC works (right button)
4) Turn off TV (Standby)
5) Turn on
6) Check no CEC works
7) Select HDMI CEC port with name "Kodi" on TV
8) check CEC works (right button)

`osmc@osmc:~$ cec-client -r
using settings from EEPROM
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: 
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [              52]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              53]	Open - vc_cec initialised
DEBUG:   [              53]	logical address changed to Free use (e)
NOTICE:  [              53]	connection opened
DEBUG:   [              54]	processor thread started
DEBUG:   [              54]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [              54]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              54]	<< e0
DEBUG:   [              85]	>> POLL sent
DEBUG:   [              85]	TV (0): device status changed into 'present'
DEBUG:   [              85]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [              85]	<< e0:8c
TRAFFIC: [             285]	>> 0f:87:00:90:3e
DEBUG:   [             285]	>> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             286]	TV (0): vendor = Philips (00903e)
DEBUG:   [             286]	expected response received (87: device vendor id)
DEBUG:   [             287]	replacing the command handler for device 'TV' (0)
NOTICE:  [             287]	registering new CEC client - v4.0.2
DEBUG:   [             287]	detecting logical address for type 'recording device'
DEBUG:   [             287]	trying logical address 'Recorder 1'
DEBUG:   [             287]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             287]	<< 11
TRAFFIC: [             707]	<< 11
DEBUG:   [             977]	>> POLL not sent
DEBUG:   [             978]	using logical address 'Recorder 1'
DEBUG:   [             978]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             978]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             978]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             978]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             978]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             978]	logical address changed to Recorder 1 (1)
DEBUG:   [             979]	Recorder 1 (1): osd name set to ''
DEBUG:   [             979]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             979]	GetPhysicalAddress - physical address = 1000
DEBUG:   [             979]	AutodetectPhysicalAddress - autodetected physical address '1000'
DEBUG:   [             979]	Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             979]	<< Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [             980]	<< 1f:84:10:00:01
NOTICE:  [            1130]	CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0,  compiled on Fri Feb  3 19:33:49 UTC 2017 by root@compiler2 on Linux 3.16.0-4-amd64 (armv7l), features: P8_USB, DRM, P8_detect, RPi
DEBUG:   [            1131]	Recorder 1 (1): vendor = Philips (00903e)
DEBUG:   [            1131]	replacing the command handler for device 'Recorder 1' (1)
DEBUG:   [            1131]	<< Recorder 1 (1) -> TV (0): OSD name ''
TRAFFIC: [            1131]	<< 10:47
DEBUG:   [            1191]	<< requesting power status of 'TV' (0)
TRAFFIC: [            1192]	<< 10:8f
TRAFFIC: [            1339]	>> 01:90:00
[... 78 lines cut ...]
DEBUG:   [          140048]	Recorder 1 (1) was already marked as active source
DEBUG:   [          140048]	>> TV (0) -> Broadcast (F): set stream path (86)
NOTICE:  [          140048]	>> source activated: Recorder 1 (1)
NOTICE:  [          140048]	<< Recorder 1 (1) -> broadcast (F): active source (1000)
TRAFFIC: [          140048]	<< 1f:82:10:00
WARNING: [          140288]	unhandled response received: opcode=82 initiator=1 destination=f response=0
TRAFFIC: [          145783]	>> 01:44:04
DEBUG:   [          145784]	SetCurrentButton right (4) D:0ms cur:4
DEBUG:   [          145784]	key pressed: right (4) current(ff) duration(0)
DEBUG:   [          145784]	Changed key right (4) D:0ms cur:ff
DEBUG:   [          145784]	key pressed: right (4, 0)
DEBUG:   [          145784]	CheckKeypressTimeout T:180.165
DEBUG:   [          145784]	Key right: idle (duration:0) (4) timeout:0ms (rel:500,rep:0,prs:500,rel:0)
DEBUG:   [          145784]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
TRAFFIC: [          146143]	>> 01:8b:04
DEBUG:   [          146143]	key released: right (4) D:359ms
DEBUG:   [          146143]	>> TV (0) -> Recorder 1 (1): vendor remote button up (8B)`

Befor update all works perfect.
I tried:
1) Power off raspberry and TV for 5 min+
2) Disable CEC on TV, reboot TV and enable
3) Delete from config hdmi_ignore_cec_init=1

Any suggestion ?

renebk, 2017-02-14:
Hi,
I can report the same problem. This happens say 9 out of 10 times. System is Lbuntu 16.04 running on an Intel N3150. Just recently upgrade to Kodi Krypton v17.0. Before the upgrade, I never experienced any sort of problem in this regard. I'veuploaded log, catching the problem. On first power cycle, the apapter reconnects and activates the connection, "CEC tester" is appearing on screen. Second cycle, nothing happens, until I reselect the HDMI input on the TV. Then the CEC adapter registers.

[cec-client.txt](https://github.com/Pulse-Eight/libcec/files/775141/cec-client.txt)

nsenica, 2017-02-16:
+1 here.. same issue here with a LG TV.

Also sometimes the remote keeps on "de-sync'ing" CEC, if inactive for more than 2 or 3 seconds. It seems to be constantly trying to sync.

Anonym-tsk, 2017-02-28:
I have same problem with PulseEight, Ubuntu 14.04.5 and Kodi v17

saperlot, 2017-03-17:
similar issues here. After going into settings of cec within Kodi (on RPI2, OSMC) sec doesn't work any more. I also the suffer the issue that the power off command that CEC should send to the TV on power down does not work. Some recent changes to the code here was not really helpful.

doe1097, 2017-03-29:
Same Problem here with LibreELEC (Krypton) v8.0.1 on a Raspberry Pi 2 in combination with a Philips  LCD. CEC-connection breaks after Standby and only re-activates after explicitly picking the RPi as input device. This wasn't a problem on Jarvis.

cmsimike, 2017-06-02:
@saperlot I am having a similar issue with Kodi - CEC on works, but not off or toggle (when its on) but cec-client works perfectly fine.

kkkkamal, 2017-06-03:
1
down vote
favorite
I need enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.

yozik04, 2017-07-11:
I have same problem on my LG TV. I need to pick Settings->Input->Simplink->HDD Recorder every time TV wakes up from sleep. Otherwise cec won't work. Everything was fine on Kodi 16

My Kodi is running on raspberry PI 2 installed via apt. I have Raspbian Jessie installed.

in /boot/config.txt on PI I have
`hdmi_force_hotplug=1`

in /home/kodi/.kodi/userdata/peripheral_data/rpi_2708_1001.xml:
```
<setting id="cec_hdmi_port" value="1" />
<setting id="cec_standby_screensaver" value="0" />
<setting id="cec_wake_screensaver" value="1" />
```

saperlot, 2017-07-11:
@cmsimike i had to get the SDCard reformatted and had to install OSMC again. I guess it's an issue related to an OSMC update..

yozik04, 2017-07-11:
Fix is here #352. It is just not merged yet.

yozik04, 2017-07-12:
I can confirm that this fix has fixed the problem on my LG TV.

doe1097, 2017-07-12:
@yozik04 Could you provide a working version? I'm using my RPi2 with LibreELEC and it seems that I would need to setup Raspbian to build it my own.

opdenkamp, 2017-07-12:
I've just merged the fix. It'll be included in the next release. Thanks

closed 2017-07-12 by opdenkamp

yozik04, 2017-07-12:
@doe1097 If you cannot wait. Here are my deb files.
https://drive.google.com/drive/folders/0B_4xizo856LmTTJFZV9HWTdaNkU
These are not signed.

Install with:
`dpkg -i <file.deb>`