Pulse-Eight/libcec #307: LG TV changes input when turned on

ashtonaut, 2017-02-12:
**Issue:**
- When the TV is turned on, after 5 seconds, it automatically switches input to the Raspberry Pi, without any user input. The Raspberry Pi is left on at all times, so this is not related to the Pi turning on.

**To reproduce:**
1. Leave RPi running
2. Switch TV to broadcast input, and put TV into standby
3. Bring TV out of standby using TV remote
4. TV will turn on, and after 5 seconds automatically switch to the RPi HDMI input

**Hardware:**
- LG 42LM6410
- Raspberry Pi 3
- HDMI connection between Pi and TV

**Software:**
- RPi running OSMC 2017.02-2
- Issue started with OSMC 2017.02-1 (the Krypton update)
- I think the issue is linked to the updated version of libcec that is bundled with Kodi Krypton

**Related forum posts:**
- [This forum post](https://discourse.osmc.tv/t/tv-auto-switches-to-osmc-when-turned-on/22266/4?u=ashtonaut) outlines the issue, and includes a [debug log](http://paste.osmc.io/jicipihopa) from OSMC
- [This forum post](https://discourse.osmc.tv/t/tv-auto-switches-to-osmc-when-turned-on/22266/19?u=ashtonaut) includes a log capturing cec-client output while reproducing the issue
- Finally, [this post ](
https://discourse.osmc.tv/t/tv-auto-switches-to-osmc-when-turned-on/22266/24?u=ashtonaut) includes a link to [this debug log](http://paste.osmc.io/iqeqociwot), this time with component-specific logging enabled for libcec

**Other thoughts**
- The OSMC devs have looked at the issue and suggest that it's something that needs to be raised here.
- There are at least 3 OSMC users reporting this issue
- The issue appears to be specific to LG TVs
- I suspect that either the Pi is sending some kind of signal it shouldn't be, or (perhaps more likely, given LG's CEC implementation) the TV is incorrectly picking up something the Pi is sending and changing input when it's not supposed to.

The WAF is at an all time low, so I'm desperate for some help here...

Thanks in advance for any thoughts or help.

ashtonaut, 2017-02-13:
Per the OSMC thread on this issue (linked above), this appears to be affecting things other than just LG TVs. The issue has been reported with a Panasonic TV and a Denon receiver as well.

popcornmix, 2017-02-14:
Possibly this issue: https://github.com/raspberrypi/firmware/issues/739
Can you try downloading updated firmware:
https://github.com/raspberrypi/firmware/blob/master/boot/start_x.elf?raw=true
https://github.com/raspberrypi/firmware/blob/master/boot/fixup_x.dat?raw=true
and replace the ones on your osmc sdcard (the directory visible when sdcard is inserted in windows)

mcobit, 2017-02-14:
With this firmware the lockup of cec when applying settings still persists.
And the switching also still occurs.

ashtonaut, 2017-02-15:
If there is value in me testing that firmware as well I'm happy to, though given mcobit's response I suspect it won't fix it. I will try when I next get a chance and report back.

popcornmix, 2017-02-15:
@ashtonaut It's worth trying. There may be more than one issue at play here.

ashtonaut, 2017-02-16:
@popcornmix I tried replacing the firmware files with the two you linked to.

Aside from some unexpected changes to the Estuary skin (main menu items I had previously turned off turned back on), no change to this issue:

- TV still auto switches to RPi HDMI input when it shouldn't
- CEC hangs and is unresponsive after selecting OK in the CEC settings dialogue

This is following several reboots of both RPi and TV, including a 60 second cold shutdown of the TV.

ashtonaut, 2017-02-21:
If anyone has any ideas for further testing, debugging, other things to try, or information to provide, please let me know.

Stane1983, 2017-02-24:
I'm not sure if it's related but Amlogic's AOCEC implementation suffers from similar issue when switching sources on TV. If you're in Kodi and CEC is active (libcec version 4.0.4; not LE/OE but Buildroot based) if you switch source from input where your device is to TV directly (not via menu but separate TV button which is basically antenna input) CEC on device instantly switches back to HDMI input. But when navigating trough sources menu on TV remote and manually select antenna input as active source it stays there.

samnazarko, 2017-02-24:
This seems to have been identified as a problem with Raspberry Pi firmware currently.

We have not seen this issue on Vero 4K (AOCEC implementation).

ashtonaut, 2017-02-25:
Oh well, good to hear it's being looked at. Thanks.

FutureSharks, 2017-03-12:
Any update on this issue? I have it too, quite annoying. Is there a corresponding issue in the raspberry pi project? The issue linked above is a separate problem I think.

ashtonaut, 2017-03-13:
I'm not sure where this issue has been picked up - perhaps @samnazarko or @popcornmix can point somewhere? I can't see anything obvious in the Raspberry Pi firmware issues list, but I'm aware there are a few different CEC issues being worked on at the moment.

mcobit, 2017-03-21:
This is definately not an issue with the firmware on raspberry as it also happens on a girmware from before the recent cec changes took place.

mcobit, 2017-03-21:
Happens since Kodi 17 update that pulled in libcece 4

anomaly256, 2017-03-30:
I am having this same issue but with a pulse8 usb adapter and kodi v17.  v16 worked fine and did not do this.  I've disabled everything in the input peripheral configuration that has anything to do with changing inputs and it still happens every time I turn my TV on (LG 60LM6450 btw)

anomaly256, 2017-04-08:
@samnazarko would mine possibly be a related problem or do you wish me to open a new issue for it?  Exact same symptoms but no rpi in the setup at all.

popcornmix, 2017-04-08:
@anomaly256 
Initially the issue was thought to possibly be either a pi firmware issue or a libcec 4 issue.
As the problem still occurs with previous pi firmware, appears when updating from libcec 3 to libcec 4 and occurs on other platforms I think we safely can say it is a libcec 4 issue, and this is the right place for your report.

anomaly256, 2017-04-08:
@popcornmix ok no worries, thanks

samnazarko, 2017-04-09:
Indeed, it looks like this issue is a problem with libCEC4 as we have seen this on other platforms now.

anomaly256, 2017-04-10:
I notice my adapter is on V4 firmware, and there appears to be no upgrader for later firmware versions available anywhere except for Windows.  Should I upgrade to V7 from a Windows machine before reporting/testing/bug hunting this problem?

mfarrell34, 2017-04-17:
I reproduced this as well and saw similar logging messages to those included in the initial post. In particular the TV is sending vendor command 0b to the raspberry pi after its power status changes to "on". For the LG TV command handler, CSLCommandHandler, this is interpreted as a reconnect request and the handler method HandleVendorCommandPowerOn is called in response. Unless I'm missing something this method call seems to be responsible for LibCEC setting the active source to the Pi which also changes the input on the TV to the port the Pi is connected on. 

Not sure if it's unexpected for the TV to be sending the reconnect request in this scenario or not, can anyone comment on this? The power status of the Recorder 1 (Pi) is already on when this command is received, so it doesn't seem that it should be changing the status back to "in transition from standby to on" and then to "on" after receiving this request from the TV.

jfjm, 2017-04-28:
Looking through the changelog:
https://github.com/Pulse-Eight/libcec/blob/master/debian/changelog.in

 * handling of LG vendor command 0x0B. Issue #256. Credits @Vollstrecker

I highly suspect it is this "fix" that actually created issue for many of us. Please revert back to the behavior prior to the "fix". It was working perfectly before.

rrolandd, 2017-04-28:
If I compile the libcec without those changes, is it possible to replace it somewhere in Libreelec?

I crosscompiled just the libcecm then created .profile file with LD_LIBRARY_PATH="/storage/lib:/usr/lib:/lib" and I placed the dynamic library there:
 425989 lrwxrwxrwx    1 root     root            17 Apr 28 13:36 libcec.so -> ./libcec.so.4.0.2
 425988 lrwxrwxrwx    1 root     root            17 Apr 28 13:36 libcec.so.4 -> ./libcec.so.4.0.2
 425986 -rwxrwxrwx    1 root     root       1161284 Apr 28 12:31 libcec.so.4.0.2

Without any luck :(

jfjm, 2017-04-29:
Did you rebuild the ld cache using ldconfig?

rrolandd, 2017-04-29:
No, I don't :( Milhouse was so kind to make us a build with reverted #256 Here you can get it: http://milhouse.libreelec.tv/builds/testing/RPi2/LibreELEC-RPi2.arm-9.0-Milhouse-20170428184648-%230427l-ga5bf4d2.tar

I will be able to try it tomorrow...

maart84, 2017-04-30:
I installed that build and my LG tv is no longer switching to the RPI when i turn my tv on!

ashtonaut, 2017-04-30:
Sounds promising! @samnazarko any chance you could create something that those of us using OSMC could test?

samnazarko, 2017-04-30:
Sure

I'll update the thread shortly

Sent from a mobile device, so please excuse the brevity of this reply

rrolandd, 2017-04-30:
I can also confirm that it solved the issue with the input change when TV turned on and also the play video issue what I had when the video was paused.

tvdbergh, 2017-04-30:
This fix seems to solve the issue for me as well.

samnazarko, 2017-05-01:
In case anyone missed my thread update earlier, see https://discourse.osmc.tv/t/tv-auto-switches-to-osmc-when-turned-on/22266/59?u=sam_nazarko

S

ashtonaut, 2017-05-01:
Sam's updated libcec for OSMC (link in previous comment) appears to have fixed this issue for me.

Thanks to all involved in finding the specific commit that appears to have introduced the issue.

anomaly256, 2017-05-09:
Confirmed that reverting #256 fixes the issue for me

anomaly256, 2017-05-16:
@popcornmix any idea if a decision has been made regarding whether to revert #256 or re-engineer it to solve this problem without unsolving the problem @vollstrecker had originally?  Although I can't replicate @vollstrecker's issue with my TV I'm happy to test or give any other info that might help

Vollstrecker, 2017-05-16:
I'm now able to read the code on a sane device, and I'm a bit confused. SLCommandHandler.cpp:189 checks, if we are the active device, before it sets itself as active source.

kkkkamal, 2017-06-03:
I need to enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.

opdenkamp, 2017-07-12:
The problems are caused by LG's CEC implementation, which breaks interop with devices that do follow the spec. I've changed this LG code in another CEC implementation, and it no longer switches inputs most of the times, but it's a completely different implementation and can't be backported quickly. I'll have to check this in the next major release.

anomaly256, 2017-07-12:
So, can we get #256 reverted until the new implementation then so the many of us without @Vollstrecker's problem can resume using libcec as we did before that change?  Or are more people than initially reported experiencing @Vollstrecker's without it?

anomaly256, 2017-07-16:
Hi @opdenkamp, I notice you pushed 1a8dc6c96a1c74d2820fea1a39f4dc8057905102 a few days ago to close #344, does this also address this issue?  If not can we get #256 reverted as well?  Thanks

anomaly256, 2017-08-01:
@opdenkamp ?

anomaly256, 2017-08-03:
I guess we can take that as a 'no'.

opdenkamp, 2017-08-03:
Haven't had time to look into this yet.

anomaly256, 2017-08-03:
Ok, thanks for responding.  Just hoping we don't have to wait for a new implementation to land since it was working for most of us prior to that commit

opdenkamp, 2017-08-03:
I'm working on a new release with fixes, but other code has priority atm. I'll get it done asap

anomaly256, 2017-08-03:
Thanks heaps

mfarrell34, 2017-08-04:
Haven't tested it yet but looking at the code it looks like it will resolve this issue. From what I saw when recreating it the only problem caused by #256 was the set active source call, and this call shouldn't be made anymore in this scenario when the 0x0b command is received.

anomaly256, 2017-08-04:
I have tested it and so have others that were reporting on this issue.  Reverting #256 has fixed the problem for us locally.

mfarrell34, 2017-08-04:
You tested after applying the changes in [1a8dc6](https://github.com/Pulse-Eight/libcec/commit/1a8dc6c96a1c74d2820fea1a39f4dc8057905102) and this issue with the TV changing input was still occurring?

anomaly256, 2017-08-04:
Oh sorry I misunderstood, I tested reverting the initial commit, I haven't had a chance to test that new one yet.  I'll try this weekend if I get time

mfarrell34, 2017-08-04:
Actually I got a chance to test it last night and my earlier post was incorrect, the input is still changing after applying the [1a8dc6c](https://github.com/Pulse-Eight/libcec/commit/1a8dc6c96a1c74d2820fea1a39f4dc8057905102) change. I'll turn on logging and reproduce it again to provide more info if that would be useful.

stedaniels, 2017-10-19:
So is the current workaround for this issue to use the build mentioned in https://github.com/Pulse-Eight/libcec/issues/307#issuecomment-298149513 ?
Cheers

anomaly256, 2017-10-19:
On Fri, 20 Oct 2017 at 7:47 am, Steve Daniels <notifications@github.com>
wrote:

> So is the current workaround for this issue to use the build mentioned in #307
> (comment)
> <https://github.com/Pulse-Eight/libcec/issues/307#issuecomment-298149513>
> ?
> Cheers
>
In my case at least, yes it was

Stan-Tastic, 2017-11-24:
The link in #307 is dead, can someone re-upload it?
I have Philips TV, connected via Denon amplituner. The issue doesn't happen always, but quite often.

felipesalomao, 2017-12-29:
Hi, i have raspbian with kodi installed.. never installed libcec manually (i think kodi installed itself).. What exactly i need to change to fix that problem (auto hdmi switch after power on lg tv)? What command use on putty or files to change using ssh? Thanks

anomaly256, 2017-12-29:
We know the fix for this, reverse that broken commit that introduced the problem.  Why hasn’t the fix been merged yet?

malard, 2017-12-29:
i am sorry but we are in a major period of development with our other products ahead of ISE (tradeshow) once that is over we will update libCEC with more improvements, including this

anomaly256, 2017-12-29:
Thanks for the update @malard, glad to hear it.  I'm a little dismayed though that 10 months after being reported a small commit that appears to have served only to introduced breakage for many people while fixing an issue for 1 hasn't simply been reverted.  Kind of scratching my head about the reasoning there :confused:

felipesalomao, 2017-12-29:
While it is not fixed, could you help solve it? I never installed libcec manually, I think kodi installed it. Always when I open kodi I see notification that cec is working.. so I does not know even installation path.. I am running raspberry pi. OS is raspbian.. do you know what files I need to change?! Thank you

anomaly256, 2017-12-29:
@felipesalomao you will need to download libcec sources and either manually undo the changes from commit 9cc4c29 or checkout the commit before that one was merged, compile your modified libcec sources on the device or cross compile for raspberry pi from a PC, then remove the version of libcec.so that Kodi installed and install your newly compiled one in its place.  If this isn't something you've done before you might need to find someone to build you a version for your hardware

ronbarr, 2018-01-27:
As a new user it's a shame that the pi is broken for LG TV users. I can build a custom version but I don't want to have to maintain this over time. 

At least the pi still works as a file server once I disconnect it from the TV.

[24 more replies]

closed 2020-10-26 by ashtonaut