Pulse-Eight/libcec #308: Tv on/off from PC

rksss, 2017-02-13:
Hello, i couldnt find a solution to this simple problem:

im using a tv (lg) as a pc monitor and i would like to have my pc (windows) turn the tv off when it goes into suspend  or when i set it up from the power settings to switch off monitor, and also turn it back on when i wake it up from suspend by moving the mouse.
i have the usb cec device from Pulse Eight and libcec installed. 
where do i go from here?

Thank you

Jooobs, 2017-02-15:
Have you enable Simplink (I.e. CEC) in the TV settings. That is the first thing to check, I have an LG TV and it has a pretty poor CEC implementation (but it is just a cheap one). Mine does not allow you to switch it off via CEC, only switch it on, which really annoys me.
I've confirmed this is the case via the LG website. 

Anyway, hopefully yours has a better CEC implementation.

rksss, 2017-02-15:
Hello, I have one of the newer models Tv, and i have enabled CEC. it even works to turn it off from the button in libcec, so i know it is activated. but how do i set it up to be turned off by the computer automatically when entering sleep/ monitor off state?

rpyka, 2017-04-28:
looks like I'm a bit late, the link is dead. I'm looking to do the same exact thing. I didn't find any uploaded code for this. Any chance you still have that installer around? Thanks!

Voodoobrew101, 2017-05-06:
opdenkamp I am also very interested in your libCEC-Standby-Monitor-installer.exe file, this why I bought the thing and I cant get it to do that. I have been trying to get it to work using task scheduler but been failing miserably.

opdenkamp, 2017-07-13:
Sorry, links have been changed by Dropbox: https://www.dropbox.com/s/zp1i9dpn5hhvfj5/libCEC-Standby-Monitor-installer.exe?dl=0

Voodoobrew101, 2017-07-13:
> Sorry, links have been changed by Dropbox: https://www.dropbox.com/s/
> zp1i9dpn5hhvfj5/libCEC-Standby-Monitor-installer.exe?dl=0

Thankyou!

Voodoobrew101, 2022-07-22:
This program works for me on windows 10 by adding it to startup wit "Start Program at Startup without UAC Prompt.exe"
But is there a executable parameter to /minimize or -hide this task on startup. Every time I reboot I must minimize it manually.