Pulse-Eight/libcec #309: Kodi Wake up from tv doesn't work

Sph1re, 2017-02-14:
Hello,

Since the update of libcec (4.0.x), it's impossible to wake up kodi by turning on the tv. 
The screen is black (the standby of the display by kodi) I am obliged to press several times on the keys of the remote control to see kodi

Is it possible to downgrade libcec? or fix this?

Anonym-tsk, 2017-02-28:
I have same problem with PulseEight, Ubuntu 14.04.5 and Kodi v17

kgeree, 2017-03-08:
same issue here on ubuntu 16.04.2 and kodi 17

opdenkamp, 2017-03-15:
This is a problem in Kodi, which doesn't disable the screensaver when you wake up the TV. I did find an issue that would have delayed power on detection on Panasonic, which will be fixed in 4.0.3. Other than that, it works as expected, after pressing a button on the remote to wake the screensaver.

Anonym-tsk, 2017-03-15:
I do not use the screensaver.
The problem is that the TV connect to HDMI1 input and does not switch to Kodi-CEC, and not connect to Pulse-Eight.

opdenkamp, 2017-03-15:
Please pastebin a log with CEC debugging enabled. I could not reproduce this problem with any of the test models that I have here after e1df683b4248f15fe22f77d08d8f7e48b779c64c

Anonym-tsk, 2017-03-15:
All messages after powering on my TV: http://pastebin.com/aWzJbu9N

opdenkamp, 2017-03-15:
CEC logging isn't enabled in system -> debugging -> log components

Anonym-tsk, 2017-03-15:
Sorry, right log - http://pastebin.com/ABjgHL3f

Anonym-tsk, 2017-03-15:
And after i select Kodi on TV, i see http://pastebin.com/ENxwchS2

opdenkamp, 2017-03-15:
I don't see anything wrong in this log. What exactly isn't working here?
I can see that you've got (at least) two source devices: Kodi on physical address 1.0.0.0 and another device on 2.0.0.0, possibly libCEC based too judging from the traffic.

Kodi on 1.0.0.0 is marking itself as active source correctly here after the TV came out of standby, so I don't see any reason for the TV not selecting the correct input and routing key presses

Anonym-tsk, 2017-03-15:
Please see the video https://youtu.be/CXH9y8JqC_8

opdenkamp, 2017-03-15:
Aha, the first log that you've uploaded was what happened after you powered up, and then the second one was after you used that source selection?

This change would explain why: https://github.com/Pulse-Eight/libcec/commit/5926317200ff78a1c03269c377c25f77f0a0f4f3

Some devices are expecting an `<active source>` command after a `<routing change>`, and some don't like it when we send it. I'll commit a work around in a few. Are you able to compile from source and test?

Anonym-tsk, 2017-03-15:
Yes, i can. Thank you.

opdenkamp, 2017-03-15:
can you try this please

opdenkamp, 2017-03-15:
oops, forgot to commit something, 2 minutes

opdenkamp, 2017-03-15:
ok fixed now, please try latest master (d1a708b9401cc20efd44f8549edffc972e4786b0)

Anonym-tsk, 2017-03-15:
Sorry, i can try it in a week only

Anonym-tsk, 2017-03-23:
Hello.
I was compiled and tested version from latest master. It works! Thank you.
Do you have a ppa with builds from master?

Wanderlei777, 2017-04-13:
I had this issue with Libreelec 8.0.1 on R-Pi 2 and LG tv simplink implementation of CEC.

After sleep or powering tv back on sometimes CEC would not wake up. Kodi itself was waking up just not CEC functionality. Changing tv or turning simplink on/off tv could resume functionality, sometimes not. 

Used compiled version with fix, from the this thread and has been working since then.

https://forum.libreelec.tv/thread-7114-page-2.html

(sorry log not enable before I moved to working version with fix)

Anonym-tsk, 2017-05-19:
I created ppa with latest version: https://launchpad.net/~anonym-tsk/+archive/ubuntu/libcec

kozo8603, 2017-05-22:
Hi Guys,

I experience the same problem: when switching to HDMI source on my TV, Kodi does not wake up. Also Kodi says libcec plugin connect failed.
I've downloaded  the latest build from this ppa, but it's still the same.
I might did something wrong during the install...
Do I need to do anything else than add this ppa and make an apt-get update?

Anonym-tsk, 2017-05-22:
@kozo8603 **apt-get upgrade** or **apt-get install libcec3**

kozo8603, 2017-05-22:
Thanks for your quick reply :)
I suspect something is not working well for me.
I might have had to start with that I have Ubuntu 16.04.2 LTS (Xenial) running on a HTPC (Intel atom based)

Since I've failed with the ppa install (always wants to pull from pulse-eight ppa) I decided to download your packages and install manually with "dpkg -i <package name>" and got the following as an example of the 5 packs :

[image]

```
htpc@user:/mnt/BackStore/_System/Ubuntu/HDMI_CEC$ sudo dpkg -i python-libcec_4.0.2.1ppa2-trusty_amd64.deb 
(Reading database ... 261095 files and directories currently installed.)
Preparing to unpack python-libcec_4.0.2.1ppa2-trusty_amd64.deb ...
Unpacking python-libcec (4.0.2.1ppa2~trusty) over (4.0.2.1ppa2~trusty) ...
Setting up python-libcec (4.0.2.1ppa2~trusty) ...

```

Anyhow the version check shows that 4.0.2.1ppa2~trusty the result is the same, Kodi can not connect to libcec.

Sorry if I seem to be a noob, but in deed I am. 😄

Wanderlei777, 2017-05-23:
I notice this peculiar behavior, even with patch applied. (Kodi Libreelec 8.0.1)

If Kodi is playing something when tv is switched off. (What I do is play music and switch off tv and listen with just speakers.) What happens when I turn tv back on, Kodi will wake up but CEC/simplink will not. It requires me to open simplink on tv and click on it and then it resumes. Normally I just turn tv on, wait a few seconds and pressing remote buttons makes it work.

Anonym-tsk, 2017-05-29:
@kozo8603 my ppa have only version for trusty. What version Ubuntu you are use?

kozo8603, 2017-05-29:
@Anonym-tsk I have Ubuntu 16.04.2 LTS (Xenial) running on a HTPC (Intel atom based)

Anonym-tsk, 2017-05-29:
@kozo8603 can you build libcec from source?

kozo8603, 2017-05-30:
@Anonym-tsk I've tried to explore how to do it, even found the method, but I'm a bit unsure on which source I'd need to download.
I'm happy to play with it you can give some guide, how to start :)
Alternatievly I also could appriciate you have some time to create a xenial versioned ppa

Thanks

Anonym-tsk, 2017-05-30:
@kozo8603 I will build a version for Xenial in my ppa today.

Anonym-tsk, 2017-05-30:
@kozo8603 Please try libcec from ppa:anonym-tsk/libcec

kozo8603, 2017-05-31:
@Anonym-tsk, You are awesome, thank you. 👍 :)
In deed after updating to your package, the TV can connect to Kodi, and the TV shows Kodi context menus.
Now just 1 thing left for me, to find a matching keymap to the RC.

Thanks again.

Anonym-tsk, 2017-05-31:
@kozo8603 maybe addon [Keymap Editor](http://kodi.wiki/view/Add-on:Keymap_Editor) help you

NeverWinter26, 2017-06-01:
Is there any way to fix this for LibreELEC (v8.0.2 MR)?
Have a Pulse-Eight (v7 firmware) running on a HTPC.

CEC works fine (from what I can tell) until I switch the TV off - it suspends Kodi but will not wake it up and I have to reset the machine.

Thanks.

Anonym-tsk, 2017-06-01:
@NeverWinter26 I wasn't work with LibreELEC. Try to build libcec from source.

kozo8603, 2017-06-01:
@Anonym-tsk 
Thank you, in deed, Keymap editor works, however, only MCE remote's signal is recognized. More likely it behaves like an Xbox controller (only navigation buttons, no play/pause button signal is processed, etc). 
I know it's going a bit behind of the topic of this thread, but do you have any insights on this?

Also, when I start Kodi from phone or remote desktop, the TV turns on But if I turn on TV (on HDMI source), Kodi does not start.

kkkkamal, 2017-06-03:
I need to enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.

opdenkamp, 2017-07-12:
Ok, so if I got this correctly, it's been fixed in the master branch, but the builds haven't been updated yet. Closing this. If I was wrong, let me know, and I'll reopen. Thanks for the builds @Anonym-tsk :)

closed 2017-07-12 by opdenkamp

Sph1re, 2017-09-22:
the initial problem has not been solved... 
when I turn on the tv, the screen remains in standby (black), I have to press a key again to see kodi. 
It seems a little innocuous, but if the tv is started during the screensaver, it will remain on indefinitely.

Sph1re, 2017-09-22:
My system:

- PC
- PulseEight (v7 firmware) 
- Kodi V17 (Krypton)
- Windows 10
- Sony TV