Pulse-Eight/libcec #313: TV device disappears

rayzorben, 2017-02-22:
Using cec-client on RPi. I turn on my TV, and do an `echo scan | cec-client -s -d 1` and get output showing my Samsung TV.

I can turn it off via `echo standby 0 | cec-client -s` and as well I can turn it on.

After using the remote control to turn off the TV (haven't tested with using cec-client) the TV continues to show up in standby mode.

After a while however, the TV disappears, and an `echo scan | cec-client -s -d 1` does not report the TV at all. At this point I am unable to turn the TV on.

opdenkamp, 2017-03-15:
Please pastebin the output, without `-d 1` or nothing of interest will show up

opdenkamp, 2017-07-12:
Closing this since there has been no response for 4 months

closed 2017-07-12 by opdenkamp