Pulse-Eight/libcec #314: Unable to build Debian package from HEAD

voltagex, 2017-02-26:
```
[100%] Building CXX object src/libcec/CMakeFiles/_cec.dir/libcecPYTHON_wrap.cxx.o
/home/voltagex/src/libcec/src/libcec/libcecPYTHON_wrap.cxx: In function 'PyObject* _wrap_new_AdapterDescriptor(PyObject*, PyObject*)':
/home/voltagex/src/libcec/src/libcec/libcecPYTHON_wrap.cxx:1165:111: warning: 'argv[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIG_Python_ConvertPtr(obj, pptr, type, flags)  SWIG_Python_ConvertPtrAndOwn(obj, pptr, type, flags, 0)
                                                                                                               ^
/home/voltagex/src/libcec/src/libcec/libcecPYTHON_wrap.cxx:8146:13: note: 'argv[0]' was declared here
   PyObject *argv[2];
             ^
Linking CXX shared module _cec.so
make[3]: Leaving directory '/home/voltagex/src/libcec'
[100%] Built target _cec
make[2]: Leaving directory '/home/voltagex/src/libcec'
Install the project...
-- Install configuration: "Release"
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/bin/./cec-client-4.0.2
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/bin/./cec-client
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/bin/./cecc-client-4.0.2
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/bin/./cecc-client
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/bin/./pyCecClient
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/lib/python3.4/site-packages/cec/_cec.so
-- Removed runtime path from "/home/voltagex/src/libcec/debian/tmp/usr/lib/python3.4/site-packages/cec/_cec.so"
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/lib/python3.4/site-packages/cec/__init__.py
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/lib/arm-linux-gnueabihf/pkgconfig/libcec.pc
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/include/libcec/cec.h
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/include/libcec/cecc.h
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/include/libcec/cecloader.h
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/include/libcec/cectypes.h
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/include/libcec/version.h
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/lib/arm-linux-gnueabihf/libcec.so.4.0.2
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/lib/arm-linux-gnueabihf/libcec.so.4
-- Installing: /home/voltagex/src/libcec/debian/tmp/usr/lib/arm-linux-gnueabihf/libcec.so
-- Removed runtime path from "/home/voltagex/src/libcec/debian/tmp/usr/lib/arm-linux-gnueabihf/libcec.so.4.0.2"
make[1]: Leaving directory '/home/voltagex/src/libcec'
   dh_install
dh_install: python-libcec missing files (usr/lib/python*/dist-packages/cec/*), aborting
debian/rules:7: recipe for target 'binary' failed
make: *** [binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2
```

Python support is enabled.

voltagex, 2017-02-26:
Aha! It seems like not every Debian distro has /etc/lsb_release, so it's failing your dist-packages vs site-packages check

mkreisl, 2017-03-03:
Ugly workaround:

`sudo touch /etc/lsb-release`

commit f4e2ed77 (2017-03-15): fixed: detect debian based distros properly when installing python. closes #314
closed 2017-03-15 by opdenkamp