Pulse-Eight/libcec #316: Python-Bindings not functional

caffeineflo, 2017-03-08:
After building and installing the latest version of libcec on my raspberry pi, I'm no longer able to use the cec python binding.
All packages that have a dependency on the cec package fail and the cec package itself refers to a _cec package that has an error as well.

Here's some of the error output
```
[root@mediapi libcec-rpi]# python3 -m cec  
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/cec/__init__.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 948, in _find_and_load_unlocked
ModuleNotFoundError: No module named '_cec'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/usr/lib/python3.6/site-packages/cec/__init__.py", line 17, in <module>
    _cec = swig_import_helper()
  File "/usr/lib/python3.6/site-packages/cec/__init__.py", line 16, in swig_import_helper
    return importlib.import_module('_cec')
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_cec'
```

#300 seems to point to the same issue, although it's unclear. 
I found several other posts regarding the same issue, so it might not just be me!

jbormolini, 2017-03-09:
I had the exact same problem with Python 3.6.0 on Arch linux.

To get it working, I modified `cec/__init__.py` line 8 to remove the >= python-2.7.0 path:
from
`if _swig_python_version_info >= (2, 7, 0):`
to
`if False and _swig_python_version_info >= (2, 7, 0):`

This is obviously a hack and to be honest I don't really understand it. I think it is a problem with swig (the thing that helps generate python to c bindings) or the way libcec is using swig. Maybe https://github.com/swig/swig/issues/769 ?

caffeineflo, 2017-03-10:
I knew I should have learned python ... but I'm sooo stuck with Swift and ObjC

Anyways, this is great! Thanks for sharing this workaround my stuff works for now, so let's hope this gets fixed soon

opdenkamp, 2017-03-15:
That is code that's being generated by Swig, which doesn't seem to be recognising your Python version. I've verified building against Python 2 and 3 and both are working, so it seems like https://github.com/swig/swig/issues/769 is causing your problems indeed.

closed 2017-03-15 by opdenkamp

caffeineflo, 2017-03-15:
So why again is this closed?
It's certainly a problem that several people reported and does affect the functionality of the python bindings to this project. So instead of closing this, wouldn't it be a solution to either keep this open and linked to the source of the problem (swig/swig#769) or have a workaround in the meantime till this is solved ?!

opdenkamp, 2017-03-15:
Because this is not a problem in libCEC or the way we use Swig or Python. It's a problem in the version of Python on your system, in combination with a bug in Swig on your system. So it's entirely external and not something we should address with some kind of hack, but something for Swig or distribution maintainers.