Pulse-Eight/libcec #317: compile for Windows x64

Rechi, 2017-03-08:
currently `src/libcec/src/cec-client/CMakeLists.txt`, `src/libcec/src/cecc-client/CMakeLists.txt` and `src/libcec/src/libcec/cmake/CheckPlatformSupport.cmake` add the definition `-D_USE_32BIT_TIME_T`. They are only added if `NOT ${WIN64}`, but `WIN64` is never defined.

There are two options
- completely remove `-D_USE_32BIT_TIME_T`
- define `WIN64` in a cmake file if target is x64

I can make a PR with the changes, but don't know which option I should take.

opdenkamp, 2017-03-15:
It took me quite some time to figure out where I set this myself... https://github.com/Pulse-Eight/libcec-support/blob/master/windows/cmake/generate.cmd#L43

If you know a good way to detect this in cmake, then a patch is very welcome, but on amd64 systems, we still have to be able to build 32 bits, which is why it's done in the batch file now

opdenkamp, 2017-07-12:
This should be ok now that your patch has been merged

closed 2017-07-12 by opdenkamp