Pulse-Eight/libcec #319: Raspberry Pi is player and recorder at the same time

PiRune, 2017-03-12:
My libcec python script (with tx commands) suddenly stopped working a few weeks ago, and it appears that it's because my Raspberry Pi is now both a recorder and player. I did a scan and got the following result:

```
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.4
power status:  on
language:      dan

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

device #4: Playback 1
address:       f.f.f.f
active source: no
vendor:        Unknown
osd string:    Playback 1
CEC version:   unknown
power status:  unknown
language:      ???

device #5: Audio
address:       4.0.0.0
active source: no
vendor:        Denon
osd string:    AVR-X3200W
CEC version:   1.4
power status:  on
language:      ???

device #8: Playback 2
address:       4.2.0.0
active source: no
vendor:        Sony
osd string:    PlayStation 4
CEC version:   1.3a
power status:  standby
language:      ???

device #B: Playback 3
address:       4.1.0.0
active source: yes
vendor:        Unknown
osd string:    Apple TV
CEC version:   1.4
power status:  on
language:      ???
```

Can anyone explain why after 1 year without issues, it suddenly decides to appear as a player and stops working?
And more importantly: **How do I prevent the raspberry from appearing as a player**? 

I checked the boot config and it still says "hdmi_ignore_cec_init=1" (and I'm running a clean Pixel install, not NOOBS).

I really hope someone can help :)

popcornmix, 2017-03-12:
What does ``vcgencmd version`` report?

PiRune, 2017-03-13:
```
Mar  3 2017 13:42:34 
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
```

PiRune, 2017-03-13:
@popcornmix I get the following:

Mar  3 2017 13:42:34 
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)

opdenkamp, 2017-03-15:
Please pastebin the full output of cec-client. It seems like you're receiving acks from an address which isn't really present. The Pi does not appear as player here, but as recorder 1.

What happens when you disconnect the other HDMI equipment?

PiRune, 2017-03-15:
I'll paste the output at my earliest opportunity, but I'm pretty sure the Pi is both recorder and player 1. 

On my tv I see the list of CEC devices, and when I unplug and turn off the raspberry, both player 1 and recorder disappeares. I've tried this multiple times to get the player to disappear.

opdenkamp, 2017-03-15:
That could be a firmware bug then, because `hdmi_ignore_cec=1` is supposed to prevent the Pi from allocating an address at boot time as far as I know, and we allocate a recorder. Something is acking on address 4 according to this, but to nothing other than that.

popcornmix, 2017-03-15:
@PiRune Power off TV from mains for a minute. I've frequently seen problems from a TV remembering state from CEC connections in the distant past that get cleared by this.
After powering on run with just the Pi connected.

Your firmware should be good. There had been some CEC firmware changes from around Christmas, but they have all been reverted since March 1st firmware.

PiRune, 2017-03-15:
Without changing anything, my script suddenly started working again. Address #4 is now Apple TV again as it should be. No weird "f.f.f.f" address. 

I hate periodic errors. Can't be sure that the problem does not occur again tomorrow.

opdenkamp, 2017-07-12:
Not sure what it was then. I'll close this ticket for now. Feel free to reopen if it occurs again

closed 2017-07-12 by opdenkamp

trinitronx, 2018-01-15:
I am also seeing a very weird issue with the `f.f.f.f` address, using Pulse8 CEC USB adapter (not on a Raspberry Pi... although I do have one running Kodi attached to same TV).  I **can reproduce** using a custom built CEC client.

I have some very "rough around the edges" [golang code][1] which can reproduce this issue in my environment.  It uses [a CGO binding][2] into official `libcec` from OS package (Ubuntu Artful `17.10`: `libcec4 4.0.1+dfsg1-1`, Linked against: `libcec-dev 4.0.1+dfsg1-1`).

Seems that after running the Golang code binary `test-cec` to transmit some CEC commands, it results in adding itself as a device with physical address `f.f.f.f`.  (Does this address even make sense??  Is this like broadcast address in TCP/IP?  Maybe this is a problem with the way that the Golang code does [`Open()`][3] [`libcec_initialise()`][4], [`getAdapter()` or `openAdapter()`][5]

I can then run `echo 'scan' | cec-client  -s`  to show that this fake device from Golang code as client is replaced by official `cec-client` (`osd string:    CECTester`).  `cec-client` seems to fix itself onto **`1.0.0.0`** address (which is already (**in use** by my Raspberry Pi).  **However**, the weird part is that the device running Golang code and subsequent `cec-client` is actually on **HDMI-4**, which should be **`4.0.0.0`** *correct*?

While playing around with this Golang code & sending the "power on" and "set active source" commands using various source & dest logical address, I somehow got things in a state where the CEC commands sent from TV remote would not be received anymore by Kodi on Raspberry Pi, and MythTV on same box as HDMI-4.  This could be some problem with the remote itself, as I usually can unplug batteries and re-plug them in and it works again.  This time it didn't.     I could then unplug TV & plug it back in... then any `f.f.f.f` device and `1.0.0.0` goes away and it is set correctly back to `4.0.0.0`.  Devices again responded.   Not sure what was going on there or how to debug it.

Either way, there's some weirdness with addressing going on... not sure how the CEC standard says this is supposed to be derived if I'm seeing `f.f.f.f` and duplicate `1.0.0.0` for the same device using my client & official `cec-client`.

@opdenkamp :  Is this something that I could ask you or another `libcec` developer to take a quick look at just to sanity check if everything looks right?  Maybe it's something simple or stupid I'm missing.  I'm a newbie to this CEC protocol & golang in general :-)

[-->Logs for both commands are here<--][6]

[1]: https://github.com/trinitronx/test-cec
[2]: https://github.com/chbmuc/cec/blob/master/libcec.go
[3]: https://github.com/chbmuc/cec/blob/master/cec.go#L58-L82
[4]: https://github.com/chbmuc/cec/blob/master/libcec.go#L53-L69
[5]: https://github.com/chbmuc/cec/blob/master/libcec.go#L71-L88
[6]: https://gist.github.com/trinitronx/cb96187c00b5077e44ea165405d078d8