Pulse-Eight/libcec #320: Adapter crashes sometimes, cec-client returns TRANSMIT_FAILED_ACK

TheNetStriker, 2017-03-13:
I'am using an "Pulse-Eight USB-CEC Adapter" on a Rasperrby Pi 3 to poll every 5 Minutes if my TV is on. (In case the power was cut from the tv) Sometimes it seams the adapter is crashing and the "cec-client" returns an "TRANSMIT_FAILED_ACK" response to the "pow" command. After a few minutes the adapter is working again and returns the correct state of the tv. Here are the logs from one of those crashes:

2017-03-12 13:53:56.714 [CecClientHandler          ] - power status: on
2017-03-12 13:53:56.714 [CecClientHandler          ] - TRAFFIC: [       135900395]	>> 01:90:00
2017-03-12 13:53:56.715 [CecClientHandler          ] - DEBUG:   [       135900395]	expected response received (90: report power status)
2017-03-12 13:53:56.715 [CecClientHandler          ] - DEBUG:   [       135900395]	>> TV (0) -> Recorder 1 (1): report power status (90)
2017-03-12 13:58:56.439 [CecClientHandler          ] - DEBUG:   [       136200120]	<< requesting power status of 'TV' (0)
2017-03-12 13:58:56.440 [CecClientHandler          ] - TRAFFIC: [       136200120]	<< 10:8f
2017-03-12 13:58:56.511 [CecClientHandler          ] - DEBUG:   [       136200192]	CEC transmission - received response - TRANSMIT_FAILED_ACK
2017-03-12 13:58:56.512 [CecClientHandler          ] - TRAFFIC: [       136200192]	<< 10:8f
2017-03-12 13:58:57.611 [CecClientHandler          ] - power status: unknown
2017-03-12 13:58:57.612 [CecClientHandler          ] - DEBUG:   [       136201293]	expected response not received (90: report power status)
2017-03-12 13:58:57.613 [CecClientHandler          ] - DEBUG:   [       136201293]	TV (0): power status changed from 'on' to 'unknown'
2017-03-12 13:58:57.613 [CecClientHandler          ] - DEBUG:   [       136201293]	TV (0): power status changed from 'on' to 'unknown'
2017-03-12 13:58:57.624 [HdmiCecHandler            ] - Cec power change from on to unknown
2017-03-12 14:03:02.239 [CecClientHandler          ] - DEBUG:   [       136445921]	command 'PING' was not acked by the controller
2017-03-12 14:03:03.740 [CecClientHandler          ] - DEBUG:   [       136447421]	command 'PING' was not acked by the controller
2017-03-12 14:03:56.458 [CecClientHandler          ] - DEBUG:   [       136500139]	<< requesting power status of 'TV' (0)
2017-03-12 14:03:56.459 [CecClientHandler          ] - TRAFFIC: [       136500139]	<< 10:8f
2017-03-12 14:03:56.680 [CecClientHandler          ] - power status: on
2017-03-12 14:03:56.681 [CecClientHandler          ] - TRAFFIC: [       136500361]	>> 01:90:00
2017-03-12 14:03:56.682 [CecClientHandler          ] - DEBUG:   [       136500361]	TV (0): power status changed from 'unknown' to 'on'
2017-03-12 14:03:56.683 [CecClientHandler          ] - DEBUG:   [       136500361]	TV (0): power status changed from 'unknown' to 'on'
2017-03-12 14:03:56.694 [HdmiCecHandler            ] - Cec power change from unknown to on
2017-03-12 14:03:56.698 [CecClientHandler          ] - DEBUG:   [       136500362]	expected response received (90: report power status)
2017-03-12 14:03:56.698 [CecClientHandler          ] - DEBUG:   [       136500362]	>> TV (0) -> Recorder 1 (1): report power status (90)

As this happened I was sitting in front of the tv and the tv was powered on the whole time. Are there any known issues with the Pulse-Eight adapter? Is there a way to solve this problem?

opdenkamp, 2017-03-15:
This isn't caused by a problem with the adapter, but with the HDMI cable or the TV, which isn't responding. The TV is not acking transmissions sent to it.

Try replacing the HDMI cable between the TV and adapter. Also, you don't really need a USB-CEC when using a Raspberry Pi like this. I can't tell from this log if libCEC is using the USB-CEC adapter here, or the Pi's internal CEC wire.

closed 2017-03-15 by opdenkamp

f1cscari, 2021-12-04:
I have the exact same Problem, but with Windows 10 - CEC works and controls for days the TV correct and suddenly, i get the
"command 'CEC transmission' was not acked by the controller" error. The fast fix is, i turn on the TV by the IR remote controller and CEC is back working again for a few days... with a google search i landed first here to find a fix but most likely, there is only a standard sellshop newbie answer, swap the HDMI cable or the TV ... 🤣