Pulse-Eight/libcec #321: rpi3 and Kodi 17 device not seen when connected to AVR (works with rpi2 and Kodi 16)

praaphorst, 2017-03-15:
Currently I have a rpi2 with Kodi 16/LibreELEC 7.0.3 connected to my AV-Receiver. When i scan for devices on my Samsung TV it finds the Kodi. Everything works fine.

Problems started when I attempted to switch to a rpi3 with Kodi 17. When connected to the AV-Receiver the scan does not see the Kodi device. First thing I did was comparing the configuration Xml with the Kodi 16 versions. They where (almost) the same, one setting changed its name.

I remembered a HDMI CEC setting where I could configure that Kodi is connected to the AVR rather then the TV. This settings seems to be gone. I am out of ideas on how to fix this. Please advice.

The current configuration is as follows:

<settings>
    <setting id="activate_source" value="0" />
    <setting id="button_release_delay_ms" value="0" />
    <setting id="button_repeat_rate_ms" value="0" />
    <setting id="cec_hdmi_port" value="1" />
    <setting id="cec_standby_screensaver" value="0" />
    <setting id="cec_wake_screensaver" value="0" />
    <setting id="connected_device" value="36037" />
    <setting id="device_name" value="Kodi" />
    <setting id="device_type" value="1" />
    <setting id="double_tap_timeout_ms" value="300" />
    <setting id="enabled" value="1" />
    <setting id="pause_or_stop_playback_on_deactivate" value="36044" />
    <setting id="pause_playback_on_deactivate" value="0" />
    <setting id="physical_address" value="0" />
    <setting id="power_avr_on_as" value="1" />
    <setting id="send_inactive_source" value="0" />
    <setting id="standby_devices" value="231" />
    <setting id="standby_devices_advanced" value="" />
    <setting id="standby_pc_on_tv_standby" value="36028" />
    <setting id="standby_tv_on_pc_standby" value="0" />
    <setting id="tv_vendor" value="0" />
    <setting id="use_tv_menu_language" value="0" />
    <setting id="wake_devices" value="231" />
    <setting id="wake_devices_advanced" value="" />
</settings>

opdenkamp, 2017-03-15:
If the setting isn't visible in Kodi, then it's auto-detected from the TV's EDID.

If Kodi doesn't show up at all, then it's usually a problem with the HDMI cable that's causing this. Try swapping it. If that doesn't help, please pastebin Kodi's debug log, with CEC logging enabled.

praaphorst, 2017-03-15:
I tried switching cables to no avail. Here’s the log:

http://sprunge.us/LQJF

Here is what I did:

Connected my pi3 directly to a samsung tv near my desktop
Activated logging
Powered down
Went to the living room
Powered down the Samsung smarttv and swapped my rpi2 for the rpi3
Powered up the tv and rpi3
Performed cec scan on tv (no devices found except for the AVR)
Shutdown power
Swapped in the rpi2
The rpi2 was immediately seen, no scan necessary

praaphorst, 2017-03-20:
Hi, this weekend I upgraded Kodi 16 on rpi2 to Kodi 17. I can now confirm that this issue is definitely tied to the version of cec supplied with that release. 

It seems to have to do with lacking a signal on start, manually switching to the correct HDMI ports does not help, there is no signal on the cable. (remember, its not the AVR or the cable, with Kodi 16 it worked fine). Renewing finds the tuner but not the Kodi.

Connecting the cable directly to the Samsung smart-TV does work, although it seems to popup as "Recorder #". 

Something is definitely wrong.

praaphorst, 2017-03-20:
More info: connect to the TV directly, renew devices on the TV and Kodi is seen. 
Reconnect to the AVR, renew devices on the TV and Kodi is seen behind the AVR.

praaphorst, 2017-03-22:
More info: Leaving the TV on and removing the power cord from the Rpi and then reconnecting it shows up a "Recorder" after renew devices on the TV. It seems to change to Kodi later on. 

I am now suspecting a timing issue during initialization ...

praaphorst, 2017-03-25:
Tried LibreElec 8.0.1, the device is now never seen when connected to the AVR, not at startup and also not after re-powering up the device.

praaphorst, 2017-03-26:
Solved the problem with hdmi_force_hotplug=1.

closed 2017-03-26 by praaphorst

opdenkamp, 2017-07-13:
Thanks for reporting back. I'll add it to the readme for the Raspberry Pi.

I've seen some recent TVs disable hotplug and EDID reading when they're in standby mode, but they still answer to CEC commands. I've seen this with recent LGs, but the symptoms that you're describing here could be from the same thing.

If there's no hotplug or if EDID reading fails, then CEC won't initialise correctly in Kodi.