Pulse-Eight/libcec #324: Panasonic TV - multimedia keys don't work, double clicks

mdodig, 2017-03-20:
Hi,
I'm using Odroid C2 with new install Libreelec 8.0, TV is Panasonic plasma ST60.
All keys on remote are working except play, pause, stop (multimedia keys don't work).
Another problem is that while navigating every click is double click.
That problem is temporarily solved by blacklisting the internal ir sensor module:
/storage/.config/modprobe.d/blacklist.conf insert "blacklist meson-ir" without quotes.
Now I'm not sure if this problem is related to ir sensor catching same command and sending it to Kodi, and that is seen as double clicks...
Please share ideas :)

Thanks,
Matko

mdodig, 2017-03-20:
Suddenly, multimedia keys started working...
Can't explain :)

mdodig, 2017-03-22:
It wasn't working with Exodus, but multimedia keys are working with local media files.
So probably a bug in addon.

9000h, 2017-04-02:
On Panasonic  to enable media control buttons on bottom of the remote, you may have to change the operation mode. To change it, press bottom *Power*-button, keep it pressed, and press *7* *3* *Stop*. After releasing *Power*-button, *Play*, *Pause*, etc should work in XBMC.
You should read your Panasonic User Manual, it's described in detail there.

opdenkamp, 2017-07-12:
Thanks for this. I've added it to the readme file.

closed 2017-07-12 by opdenkamp