Pulse-Eight/libcec #326: Panasonic - some inputs repeated

exzemat, 2017-03-23:
Hello,

I use since 3 years pulse eight to command kodi (kodibuntu, based on ubuntu 14.04 and Isengard), via hdmi-cec, with my Logitech Harmony 350 Control without problem.
I upgrade successfully to Jarvis last year, no problem with hdmi-cec.

But I upgrade to Krypton last week (and upgrade to ubuntu 16.04 because vaapi/vdapu), and now my harmony repeated some inputs : no systematically, but alway, maybe 1 per 5 or 6 inputs.
for example (look just before thumbs-up)
https://drive.google.com/file/d/0B2zHcuM...sp=sharing

I tested on ubuntu 16.04, 16.10 and with librelec (last edition with krypton), with estuary skin and mimic, and I tried some manipulation :

https://github.com/osmc/osmc/issues/174
or
https://gitlab.com/ember-dev/kodi/commit...24a65f1632

I try with the TV's remote control : same bug
I try with my phone (which have IR blaster) : no bug !

My TV is panasonic TX-P42V20E.

but without success.

benjaminr-ps, 2017-04-29:
Same for me, since I have upgraded to Krypton I'm experiencing the same behaviour with my TV.
But I think it is independent on the TV.

As you already said, It occurs occasionally, when any key of remote control is pressed 2 times, Kodi interprets it as 3 times.

For debugging, I enabled the verbose logging for CEC, to see if 2 key presses are received 3 times.
As I understand correctly the log, the 2 key presses are interpreted as 3 key presses.

I have attached the log file, thus you can verify I have not misinterpreted the logging.
[cec_log.txt](https://github.com/Pulse-Eight/libcec/files/966132/cec_log.txt)

My CEC configuration:
Remote button press delay before repeating (ms): 1000
Remote button press repeat rate (ms): 250
Remote button press release time (ms): 0

Testing:
Started TV with Kodi. Enabled CEC verbose logging. Sitting beside TV with Laptop and having tail -f -n 50 executed on .kodi/temp/kodi.log.

Any suggestions?

cecbug, 2017-05-23:
I'm seeing this bug after upgrading to libcec 3.1.0 or 4.0.2 from 3.0.1 which was working fine. Firmware 5. From changelog libcec 3.1.0:

* skip double press removal. It is handled through other means. credits @popcornmix

What other means?

Anyone on firmware 7 with this bug? Where is the firmware changelog?

gdave321, 2017-06-03:
I reverted this "skip double press removal" commit on 4.0.2 and re-built it.  Not more double presses :-)
So until these "other means" have been implemented (how?) I'm back to normal operation.

exzemat, 2017-06-23:
so...
How noobs can revert ?

benjaminr-ps, 2017-06-26:
@gdave321 Didn't you got some conflicts, when reverting to the recent source-code version.

I got some by doing the revert, so could you share a patch or make pull request?

ChristianCiach, 2017-06-29:
Can confirm. Annoying double presses in Kodi 17 on Arch Linux using the USB adapter with a Sony TV (2017 model).

gdave321, 2017-06-30:
This is patch I used on 4.0.2 to revert the change.
[https://gist.github.com/gdave321/26a3dd16af1ca990dda7589d9640d9e8](url)
could be white-space damaged so apply with patch --ignore-whitespace

ChristianCiach, 2017-06-30:
I could be wrong, but I think it's fixed for me after updating the adapter to firmware v7. I need to test is more, though.

benjaminr-ps, 2017-07-03:
@gdave321 thanks for sharing, also works for me!

ChristianCiach, 2017-07-05:
I was wrong, updating the firmware to v7 did not fix this issue. Extremely annoying..

ChristianCiach, 2017-07-08:
@gdave321 Thanks, reverting commit ef8bc8e1ffede6fc7c95c7ccf1459f1ca4df0d2c also worked for me on Arch Linux.

opdenkamp, 2017-07-12:
The firmware version doesn't affect key presses, but only low level things like bit timings and waking up the PC from standby/off.

Can you try this (response to another ticket about this):
```
Can you try to stop Kodi, remove ~/.kodi/userdata/peripheral_data, install libcec without the change reverted and then try Kodi again. I think that it may be caused by an old value of the settings.
```

ChristianCiach, 2017-07-12:
Deleted peripheral_data and reinstalled libcec: double taps are back immediately. The xml file inside peripheral_data was not recreated by kodi, so the issue happens with default settings without ever changing any peripheral settings inside Kodi.

ChristianCiach, 2017-07-12:
I want to add that double presses don't happen on every key press, just on any fifth press (or so) in average.

opdenkamp, 2017-07-13:
The xml in peripheral_data is created by Kodi :)
Could you enable debug logging in Kodi and enable component logging for CEC. Then reproduce the issue and upload (another) log. The link from the first post no longer seems to work.

ChristianCiach, 2017-07-13:
I can do so on the weekend.

ChristianCiach, 2017-07-17:
Sorry, the weekend was shorter than expected. I will come back to you :)

ChristianCiach, 2017-07-19:
Here you go:

[cec_debug.log.txt](https://github.com/Pulse-Eight/libcec/files/1160324/cec_debug.log.txt)

Included is the startup of Kodi, including the initializing of libcec. 

At line 1720 I cut out a bunch of unimportant lines. Starting at that line until the end of the log I pressed a button (up or down) on my IR remote pretty much exactly every 3 seconds, using a stopwatch. As you can see, sometimes libcec recognized two presses in one second. My remote is very clicky, a lot like mouse buttons, so there is no way that I actually accidentaly pressed a button two times in one second. The very last recognized key press in the attached log file is one of these falsely recognized double presses.

Whenever you see `received key a6 duration xxx` in my log file, where "xxx" is greater than zero, then this is a falsely recognized double press. Every correctly interpreted single press has a duration of 0 inside this log file.

ChristianCiach, 2017-07-19:
Maybe this is related: Whenever there is a noticable delay between me pressing the button and kodi performing the action, then there is a very high change that the next button press is wrongly recognized as a double press. This delay between pressing a button and kodi performing the action happens pretty frequently.

Also, when I hold down a direction button on my remote (to scroll through a list), the scrolling speed is painfully slow and also very erratic, meaning that the menu items don't change the selection on a fixed interval. These three issues (double presses, slow scrolling and erratic scrolling) are the reason why I almost only control Kodi using my smartphone.

ChristianCiach, 2017-07-19:
I wonder why the reverted "double press removal"-code was even necessary in the first place. I can't imagine my TV to actually send multiple button presses over CEC when I press the button just once. No other CEC controlled device (like the PlayStation 3 and 4) have this issue. 

I think the reverted "double press removal"-code was probably just a workaround for another bug that results in libcec (or the firmware of the adapter) recognizing multiple button presses where is actually only one.

exzemat, 2017-07-30:
@gdave321
Tour patch was removed.
Gould you upload it again please ?

satmandu, 2018-03-02:
This is the patch against master that seemed to fix the problem for me, as per the aforementioned commit reversion:

```
--- libcec-master/src/libcec/CECClient.cpp	2018-03-02 13:47:13.948464097 -0500
+++ libcec-fixed/src/libcec/CECClient.cpp	2018-03-02 13:51:30.416351490 -0500
@@ -60,8 +60,11 @@
     m_releaseButtontime(0),
     m_pressedButtoncount(0),
     m_releasedButtoncount(0),
-    m_iPreventForwardingPowerOffCommand(0)
+    m_iPreventForwardingPowerOffCommand(0),
+    m_iLastKeypressTime(0)
 {
+  m_lastKeypress.keycode = CEC_USER_CONTROL_CODE_UNKNOWN;
+  m_lastKeypress.duration = 0;
   m_configuration.Clear();
   // set the initial configuration
   SetConfiguration(configuration);
@@ -1650,7 +1653,20 @@
 {
   CLockObject lock(m_cbMutex);
   if (m_configuration.callbacks && !!m_configuration.callbacks->keyPress)
+  {
+    // prevent double taps
+    int64_t now = GetTimeMs();
+    if (m_lastKeypress.keycode != key.keycode ||
+        key.duration > 0 ||
+        now - m_iLastKeypressTime >= DoubleTapTimeoutMS())
+    {
+      // no double tap
+      if (key.duration == 0)
+        m_iLastKeypressTime = now;
+      m_lastKeypress = key;
       m_configuration.callbacks->keyPress(m_configuration.callbackParam, &key);
+    }
+  }
 }
 
 void CCECClient::CallbackAddLog(const cec_log_message_cpp &message)
```
```
--- libcec-master/src/libcec/CECClient.h      2018-03-02 13:47:13.948464097 -0500
+++ libcec-fixed/src/libcec/CECClient.h       2018-03-02 13:50:28.639232130 -0500
@@ -459,6 +459,8 @@
     int32_t                                  m_pressedButtoncount;                /**< the number of times a button released message has been seen for this press. */
     int32_t                                  m_releasedButtoncount;               /**< the number of times a button pressed message has been seen for this press. */
     int64_t                                  m_iPreventForwardingPowerOffCommand; /**< prevent forwarding standby commands until this time */
+    int64_t               m_iLastKeypressTime;                 /**< last time a key press was sent to the client */
+    cec_keypress          m_lastKeypress;                      /**< the last key press that was sent to the client */
     P8PLATFORM::SyncedBuffer<CCallbackWrap*> m_callbackCalls;
   };
 }
```

nomaddamon, 2018-03-20:
Same here on 2017 Philips OLED - Kodi 16 works fine, double presses on Kodi 17 (tested on Pi3, s905, s912).
Logs look the same as ChristianCiach posted - two CEC events very close by, one with 0ms duration, other with some duration >0. 2013 Philips LED TV works fine with Kodi 17 (same devices)

ChristianCiach, 2018-05-13:
So, will something be done about this? Seeing that this project didn't have a single commit in almost a year, I fear this issue will never be fixed.

In any way, the fix is trivial, as just one commit needs to be reverted. The commit in question dind't even try to fix anything, so there is no risk to re-introduce other bugs. What is the blocker here? Do you need more data? Should I submit the revert of the commit as a pull request?

ChristianCiach, 2018-05-15:
So, after studying quite some source code, I am not too sure that libcec is really at fault here. It could actually an issue with Kodi, but I am not certain yet.

Anyway, it looks like @popcornmix actually committed a fix/workaround to Kodi that should solve this issue for us: https://github.com/xbmc/xbmc/commit/a4a87e9c788ad3664cc5b9628cd11381fda4cf19

By tuning these values, we should be able to get these damn double presses under control. Now we just need to wait for Kodi 18 for these settings to become available :)

benjaminr-ps, 2018-05-16:
Already tried different mutations of these value without success.
But maybe you are more lucky to find the proper values.

clacor, 2018-06-09:
I have the same issue. No mutations of the 3 values seems to fix the problem...
Would like to test the suggested revert but I don't know how...
Can somebody give me a hint how to revert https://github.com/Pulse-Eight/libcec/commit/ef8bc8e1ffede6fc7c95c7ccf1459f1ca4df0d2c on Libreelec on a Raspi2?

clacor, 2019-08-19:
Just wanted to report back that the patch from @satmandu solved the issue for me on a Philips TV with libcec 4.0.4 under LibreELEC 9.2 on a Raspi2!

pozsothereal, 2020-02-21:
Hi All!

Sorry for noob question, but i have exactly the same problem with my pana viera and latest kodi.
I would try satmanu's patch but i dont really follow how it can be applyed. Please someone be so kind and mention some direction where do i start. Thank you!!

satmandu, 2020-02-21:
@pozsothereal For what it is worth, I'm not having these issues any more using current ubuntu and kodi installed from the [Kodi Stable PPA (ppa:team-xbmc/ppa)](https://launchpad.net/~team-xbmc/+archive/ubuntu/ppa) with our Panasonic tv.

pozsothereal, 2020-02-21:
> @pozsothereal For what it is worth, I'm not having these issues any more using current ubuntu and kodi installed from the [Kodi Stable PPA (ppa:team-xbmc/ppa)](https://launchpad.net/~team-xbmc/+archive/ubuntu/ppa) with our Panasonic tv.

Hi, thanks for prompt reply. Im running rpi4 w buster and kodi 17.4, and would like to control gpio pins with remote button press. Now im able to do this with rpi.gpio and xdotool via cec client, but when i press a button once always send double keypress, doesnt help when i just really try push it shortly:(

pozsothereal, 2020-02-21:
Here is a screenshot, and version:
libCEC version: 4.0.4, compiled on Linux-4.15.0-48-generic ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, AOCEC
[image]

satmandu, 2020-02-21:
@pozsothereal You can copy those patches to a file and then go to your libcec folder and run ```patch -p1 < file.patch```

That should apply the patch. (Or something similar.)

satmandu, 2020-02-21:
Note also that you're running an older version of kodi... and if this is a kodi issue you're running into, you would be better off running the curent version, which is 18.5 or so...

pozsothereal, 2020-02-22:
> Note also that you're running an older version of kodi... and if this is a kodi issue you're running into, you would be better off running the current version, which is 18.5 or so...

Ah, i just realized that cec build in Kodi -btw now im using 18.5 and problem still exist-.
So is there a folder for cec inside kodi somewhere? or i have to disable and build separately?
Thx!

satmandu, 2020-02-22:
You might want to ask on the kodi forums about this. They'll be more familiar with supporting custom kodi installs.

Forage, 2021-08-24:
I've got the same issue with a Panasonic TX-P42G30 in combination with a RPi 4 using LibreELEC 10.0 RC1 (Kodi 19.1).

@popcornmix even though it's been quite some time, can you explain the need for the change? Are the "other means" of https://github.com/popcornmix/libcec/commit/ef8bc8e1ffede6fc7c95c7ccf1459f1ca4df0d2c the three added Kodi remote parameters? If yes, to what should they be set to achieve the same?

Forage, 2025-02-27:
Wow, hardly a day to even check if the issue is still present in a more recent release and the issue already gets closed.
Yes, it's still present in 4.0.7. I keep building each LibreElec release myself to revert the breaking changes in libcec.
But as this is the version still included in LibreElec, I did not yet have the time to compile a version with 7.0.0.
It is unclear whether libcec can be replaced with a more recent version just like that, but I'll give it a go. I have the impression that Kodi at least no longer imposes 4.0.7 specifically. Though it won't be for another two weeks before I have time to test, unfortunately.

opdenkamp, 2025-02-27:
I'll reopen this one until you've had time to test it. I'm just cleaning up some older tickets now but if something is still relevant, then you can just ping in that ticket that got closed.

Forage, 2025-03-30:
With a bit more delay than anticipated, I can confirm that the issue unfortunately remains present in LibCEC 7.0.0.

My LibreElec 12.0.2 RPi 4 build with an updated LibCEC can be found here if anyone else wants to check: https://drive.proton.me/urls/TX2Y69SDP0#1mckCAfE2ikO

Sticking to reverting https://github.com/popcornmix/libcec/commit/ef8bc8e1ffede6fc7c95c7ccf1459f1ca4df0d2c for each LE release again.

clacor, 2025-03-30:
> With a bit more delay than anticipated, I can confirm that the issue unfortunately remains present in LibCEC 7.0.0.
> 
> My LibreElec 12.0.2 RPi 4 build with an updated LibCEC can be found here if anyone else wants to check: https://drive.proton.me/urls/TX2Y69SDP0#1mckCAfE2ikO
> 
> Sticking to reverting [popcornmix@ef8bc8e](https://github.com/popcornmix/libcec/commit/ef8bc8e1ffede6fc7c95c7ccf1459f1ca4df0d2c) for each LE release again.

I can confirm this as well!

closed 2025-06-29 by github-actions[bot]