Pulse-Eight/libcec #329: _cec.so should be installed in one directory higher

10robinho, 2017-04-01:
Hello,

Currently `_cec.so` installs in `site-packages/cec/` while it should be placed one level higher, in `site-packages`.

Here is simple patch for it, against current master:

```
diff --git a/src/libcec/cmake/CheckPlatformSupport.cmake b/src/libcec/cmake/CheckPlatformSupport.cmake
index 962607e..ba43db5 100644
--- a/src/libcec/cmake/CheckPlatformSupport.cmake
+++ b/src/libcec/cmake/CheckPlatformSupport.cmake
@@ -207,7 +207,7 @@ else()
       endif()
 
       install(TARGETS     ${SWIG_MODULE_cec_REAL_NAME}
-              DESTINATION lib/python${PYTHON_VERSION}/${PYTHON_PKG_DIR}/cec)
+              DESTINATION lib/python${PYTHON_VERSION}/${PYTHON_PKG_DIR})
       install(FILES       ${CMAKE_BINARY_DIR}/src/libcec/cec.py
               DESTINATION lib/python${PYTHON_VERSION}/${PYTHON_PKG_DIR}/cec
               RENAME      __init__.py)

```

ramunasd, 2017-04-05:
I've tried without this patch and everything works fine with python 3.4.
Are You experiencing any issues due to paths?

10robinho, 2017-04-05:
I use Python 3.5, maybe it makes difference. I didn't change python default settings and paths. All other modules that have shared libraries (.so) files have them placed in site-packages directory.

Anyway, you can close this if it works for you. If someone meets the same problem I hope that he can find the answer here.

opdenkamp, 2017-07-12:
I'll have to check updating Python in the next major release. Too much risk of breaking existing things when doing it in a minor release.

opdenkamp, 2017-07-18:
I've just checked this with Python 2.7 on Windows, for EventGhost #357, and it does indeed expect `_cec.pyd` to be one directory up there.

I'll check what's going on on Linux with this directory. It's possible that I picked the wrong directory here because Debian(based) systems don't always use the same structures as upstream.

opdenkamp, 2025-02-26:
fixed in libCEC 7.0.0

closed 2025-02-26 by opdenkamp