Pulse-Eight/libcec #331: Segmentation fault: 11

gingerbeardman, 2017-04-04:
Using latest Mac version of libcec/cec-client

when using monitor mode after a shirt while operating my physical devices (switch input, routing change)

`MacBookPro-Matt:libcec matt$ ./cec-client -m
starting a monitor-only client. use 'mon 0' to switch to normal mode
No device type given. Using 'recording device'
CEC Parser created - libCEC version 2.1.1
no serial port given. trying autodetect: 
 path:     /dev/cu.usbmodemv1
 com port: /dev/cu.usbmodemv1

opening a connection to the CEC adapter...
DEBUG:   [               0]	unregistering all CEC clients
DEBUG:   [               0]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [               4]	connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [               4]	communication thread started
DEBUG:   [              76]	turning controlled mode on
NOTICE:  [             100]	connection opened
waiting for input
DEBUG:   [             100]	processor thread started
TRAFFIC: [            3395]	>> 5f:80:11:00:12:00
DEBUG:   [            3395]	>> Audio (5) -> Broadcast (F): routing change (80)
DEBUG:   [            3395]	Audio (5): power status changed from 'unknown' to 'on'
DEBUG:   [            3395]	device Audio (5) status changed to present after command routing change
TRAFFIC: [            6342]	>> 5f:80:12:00:11:00
DEBUG:   [            6342]	>> Audio (5) -> Broadcast (F): routing change (80)
TRAFFIC: [           13463]	>> 50
DEBUG:   [           13463]	<< POLL: Audio (5) -> TV (0)
TRAFFIC: [           13507]	>> 50
DEBUG:   [           13507]	<< POLL: Audio (5) -> TV (0)
TRAFFIC: [           13566]	>> 5e
DEBUG:   [           13566]	<< POLL: Audio (5) -> Free use (e)
TRAFFIC: [           13618]	>> 5e
DEBUG:   [           13618]	<< POLL: Audio (5) -> Free use (e)
TRAFFIC: [           13771]	>> 5f:84:10:00:05
DEBUG:   [           13771]	>> Audio (5) -> Broadcast (F): report physical address (84)
DEBUG:   [           13771]	Audio (5): physical address changed from ffff to 1000
TRAFFIC: [           13910]	>> 5f:87:00:09:b0
DEBUG:   [           13910]	>> Audio (5) -> Broadcast (F): device vendor id (87)
DEBUG:   [           13910]	Audio (5): vendor = Onkyo (0009b0)
TRAFFIC: [           13988]	>> 5f:85
DEBUG:   [           13988]	>> Audio (5) -> Broadcast (F): request active source (85)
DEBUG:   [           13988]	>> 5 requests active source
Segmentation fault: 11
MacBookPro-Matt:libcec matt$ 
`

opdenkamp, 2017-07-12:
could you run cec-client in gdb and type `thread apply all bt` when it crashes again please. Then show me the output

radokristof, 2019-02-25:
I have some similar problem.
Latest version installed on Raspbian and when I try `echo on 0 | cec-client -s -d 1` or `echo standby 0 | cec-client -s -d 1` it hangs for 10-15 seconds then it shuts down the TV or turns on, but I immediately get a segmentation fault.

MaximusBaton, 2019-03-23:
same for my hisense

kraken22, 2020-07-28:
I'm having the same issue. libCEC version 4.0.4. Hisense TV. Raspberry pi 4. 
I am using a micro HDMI to full-size HDMI adapter.
However changing the port from port 1 to port 0 (the one closer to the power cable) fixed this issue.

opdenkamp, 2020-07-28:
please try again using the latest libCEC release (6.0.2)

audas, 2020-11-08:
How would I go about updating libCEC on the Raspberry Pi. I am testing on the new kernel 5.9 and have this segmentation fault.

gingerbeardman, 2025-02-27:
Stale after 5 years LOL

opdenkamp, 2025-02-27:
if you can still replicate this issue with libCEC 7.0.0+ then this ticket can be reopened

closed 2025-02-27 by opdenkamp