Pulse-Eight/libcec #333: odd string truncated

gingerbeardman, 2017-04-05:
`
device #4: Playback 1
address:       1.1.0.0
active source: yes
vendor:        Unknown
osd string:    NintendoSwitc
CEC version:   1.3a
power status:  on
language:      ???
`

on my TV via CEC it shows as NintendoSwitch, with the final character h

pfiaux, 2017-06-28:
It might be a limitation on the OSD string length in libcec. If the OSD string is handled the same way as `strDeviceName` in `libcec_configuration` it looks like it's limited to a length of 13 characters.

For example in cecetypes.h:
`typedef char cec_osd_name[14]; /**< the name of the device */`
(I'm not sure but my guess is 13 + null termination)

opdenkamp, 2017-07-12:
Just checked, it looks like this is a bug that has been corrected incorrectly last time... Maximum length is 14 bytes according to spec, not 13. Because other apps depend on the NULL terminator most likely, I can't change this without bumping the major version number.

opdenkamp, 2020-04-27:
this will be fixed after the version bump

commit b6f8e693 (2020-04-27): fixed: osd name truncated (after v5 bump)

closes #333
closed 2020-04-27 by opdenkamp