Pulse-Eight/libcec #334: Updated Mac build?

gingerbeardman, 2017-04-05:
I have tried to build the Mac version but it is beyond me as I do not know how to setup p8-platform

Could you please release an updated Mac build?

Thanks!

gingerbeardman, 2017-04-07:
I see v4.0.2 has just been released for Windows

Linux is all but abandoned.
Mac seems to be also.

Please help us non-Windows users.

malard, 2017-04-07:
Not really?, Mac we don't have the ability to compile. But Linux has packages within the various distros package managers, so we don't publish a download on our website https://packages.debian.org/source/sid/libcec

gingerbeardman, 2017-04-07:
Sorry I was specifically taking about the firmware upgrader.

If you give me help I'll be happy to compile the Mac version for you.

jordanade, 2017-05-26:
Yeah, where is cec-firmware-upgrade? Can anyone help?

gingerbeardman, 2017-05-28:
I ended up using VirtualBox running Windows just to upgrade my firmware.

opdenkamp, 2017-07-12:
Unfortunately we can't support firmware updates on non-Windows at the moment, but you shouldn't have to update the firmware unless you have a very old device.

closed 2017-07-12 by opdenkamp