Pulse-Eight/libcec #335: Firmware v7 reports that v4 has been installed

gingerbeardman, 2017-04-07:
Firmware upgrader reports that V4 has been installed after successful install.

Checking adapter V7 has actually been flashed.

Windows updater

[image]

opdenkamp, 2017-04-07:
This is a bug in the firmware upgrade tool, which contains a wrong label. It did actually flash the correct version of the firmware. You can verify this with "cec-client.exe -l"

gingerbeardman, 2017-04-07:
@opdenkamp that's what I meant by my second sentence

if I should report this elsewhere please let me know where exactly.

opdenkamp, 2020-04-27:
This has been fixed in more recent versions, but I forgot to close this ticket. Thanks

closed 2020-04-27 by opdenkamp