Pulse-Eight/libcec #342: CEC unavailable, cannot wake KODI from dimmed state

nuwonda, 2017-04-15:
This commit seems to be in part breaking CEC on RPI2/3 with e.g. Panasonic TVs:
https://github.com/Pulse-Eight/libcec/commit/67d444dd944192a31ab07637a4b57d27c890ede4

After some time of inactivity (KODI dimmed by screensaver) TV remote no longer works and you have to use the KODI phone app to wake up KODI after which the TV remote starts to work again.

CEC in Build 314 works
CEC in Build 315 does not work after KODI dimmed
see: http://forum.kodi.tv/showthread.php?tid=298461&page=97

VimesUK, 2017-05-24:
This also seems to cause the issue of our Panasonic TV not switching to source when the Pi3 is powered up, again this started with build 315.

One of my posts noted this here....

http://forum.kodi.tv/showthread.php?tid=298461&pid=2590588#pid2590588

opdenkamp, 2017-07-12:
This issue should be fixed in master now. Please let me know if it isn't

closed 2017-07-12 by opdenkamp

nuwonda, 2017-08-30:
I still have this issue in #829 (and earlier builds)

opdenkamp, 2017-08-30:
the code from the master branch hasn't been tagged as release yet. I'll hopefully have time to get it finished next week

nuwonda, 2017-08-30:
ahh ok, thanks for the clarification