Pulse-Eight/libcec #344: LG TV - issues with LIBCEC 4 (Source switch when TV switched on/auto play paused video)

rrolandd, 2017-04-23:
I have two issue with my LG LM640S television when using it with latest KODI 18 testbuilds. I tried and the same issue happens on the stable Libreelec and I can confirm that everything was fine with libcec 3.

My config:
TV: LG LM640S
HDMI 1: RPI2 with KODI with testbuilds from http://forum.kodi.tv/showthread.php?tid=298461
CEC adapter settings: [https://www.dropbox.com/s/gfbn4f6put60ej7/IMG_5327.JPG?dl=0](CEC adapter config in KODI)

## ISSUE 1
- RPI is always switched on from a wall adapter
When we are turning on the TV, it switch to the RPI as an active source. It does not matter if the last input was tv or not.
Same happens if tv is turned on and I restart the RPI or plug out and then plugin in RPI.

LOG: [https://www.dropbox.com/s/q9wg9yzxwgw4plh/kodi.log2?dl=0](https://www.dropbox.com/s/q9wg9yzxwgw4plh/kodi.log2?dl=0)
The log snapshot were made just after the tv auto switched the source to RPI, so the CEC log is in the last minute.

## ISSUE 2
When I pause a video with tv remote, the video starts to play after 10-20 seconds. When logging disabled, this issue comes almost every time. When logging enabled, it happens 1 in 5 times. Very annoying :)

Part of the log, where I pause the video on CEC and when it start to play: [https://pastebin.com/G8yYxNK8](https://pastebin.com/G8yYxNK8)
Full LOG: [https://www.dropbox.com/s/gif5pt15fks1hy3/kodi.log3?dl=0](https://www.dropbox.com/s/gif5pt15fks1hy3/kodi.log3?dl=0)

ashtonaut, 2017-04-23:
Your Issue #1 sounds like it is this:

https://github.com/Pulse-Eight/libcec/issues/307

This is currently unresolved.

rrolandd, 2017-04-23:
Yes, that issue sounds similar. Maybe my log files help to the developer to figure out a solution.

opdenkamp, 2017-07-12:
can you check whether this solve your second issue https://github.com/Pulse-Eight/libcec/commit/1a8dc6c96a1c74d2820fea1a39f4dc8057905102

closed 2025-02-27 by github-actions[bot]