Pulse-Eight/libcec #345: NUC Gen 7 support?

SpaceAgeHero, 2017-04-25:
Yesterday I installed Arch-Linux (as well as Kodi and libcec) on my brand new NUC7i5BNK.
Unfortunately Kodi doesn't detect a CEC remote. My TV clearly recognizes the NUC as CEC-capable.

I assume it is not yet supported? If so, can you please add support?

Sorry if my assumption is wrong. Any help appreciated.

Kind regards

opdenkamp, 2017-07-12:
We've checked the 6th generation NUC, but don't have any 7th generation here. The latest versions of the NUCs have an integrated partial CEC implementation in UEFI (BIOS), which might be what your TV is talking to. Did you disable it?

Can you show me the output of `lsusb` on your system.

9000h, 2017-07-12:
there is something coming to the intel gfx
https://patchwork.freedesktop.org/project/intel-gfx/patches/?submitter=15857

opdenkamp, 2017-07-13:
I guess that will use the new in-kernel CEC framework, which libCEC doesn't support yet.
So it will probably work with vendors who implement the CEC spec to the letter out of the box, but lots of vendors don't do that.

oferchen, 2019-05-02:
I'm trying to get LibreElec 9.1 Millhouse to work with NUC7i3DNHE
Kodi doesn't detect my TV remote and I don't think connection is being established

```
LibreELEC:~ # echo 'scan' | cec-client -s /dev/ttyACM0 -d
== skipped log-level parameter: no level given ==
opening a connection to the CEC adapter...
DEBUG:   [               0]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [               0]	connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [               6]	communication thread started
DEBUG:   [             117]	turning controlled mode on
NOTICE:  [             289]	connection opened
DEBUG:   [             289]	processor thread started
DEBUG:   [             289]	<< Broadcast (F) -> TV (0): POLL
TRAFFIC: [             289]	<< f0
DEBUG:   [             289]	setting the line timeout to 3
DEBUG:   [             429]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             429]	<< f0
DEBUG:   [             514]	CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             514]	>> POLL not sent
DEBUG:   [             514]	TV (0): device status changed into 'not present'
NOTICE:  [             514]	registering new CEC client - v4.0.4
DEBUG:   [             514]	detecting logical address for type 'recording device'
DEBUG:   [             514]	trying logical address 'Recorder 1'
DEBUG:   [             514]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             514]	<< 11
DEBUG:   [             599]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             599]	<< 11
DEBUG:   [             684]	CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             684]	>> POLL not sent
DEBUG:   [             684]	using logical address 'Recorder 1'
DEBUG:   [             684]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             684]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             684]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             685]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             685]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             685]	setting ackmask to  2
DEBUG:   [             739]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             739]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             739]	GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [             743]	GetPhysicalAddress - ADL returned physical address 0000
DEBUG:   [             743]	GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG:   [             743]	GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG:   [             743]	GetPhysicalAddress - trying to get the physical address via drm files
DEBUG:   [             743]	GetPhysicalAddress - drm files returned physical address 1000
DEBUG:   [             743]	AutodetectPhysicalAddress - autodetected physical address '1000'
DEBUG:   [             743]	Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             743]	<< Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [             743]	<< 1f:84:10:00:01
DEBUG:   [             965]	using persisted autonomous mode setting: 'disabled'
DEBUG:   [            1021]	using persisted CEC version setting: '1.4'
DEBUG:   [            1076]	using persisted logical address setting: 'Recorder 1'
DEBUG:   [            1131]	using persisted device type setting: 'recording device'
DEBUG:   [            1137]	using persisted logical address mask setting: '206'
DEBUG:   [            1203]	using persisted physical address setting: '1000'
DEBUG:   [            1203]	setting the OSD name to CECTester (previous: Kodi)
NOTICE:  [            1258]	CEC client registered: libCEC version = 4.0.4, client version = 4.0.4, firmware version = 8, firmware build date: Fri Jul 21 09:08:24 2017 +0000, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0,  compiled on Thu Feb 21 12:07:56 UTC 2019 by jenkins@prometheus on Linux 4.15.0-43-generic (x86_64), features: P8_USB, DRM, P8_detect, randr
DEBUG:   [            1258]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG:   [            1258]	<< Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [            1258]	<< 10
DEBUG:   [            1343]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [            1343]	<< 10
DEBUG:   [            1428]	CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [            1428]	>> POLL not sent
DEBUG:   [            1429]	not sending command 'set osd name': destination device 'TV' marked as not present
DEBUG:   [            1429]	<< requesting power status of 'TV' (0)
DEBUG:   [            1429]	<< Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [            1429]	<< 10
DEBUG:   [            1514]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [            1514]	<< 10
[... 123 lines cut ...]
DEBUG:   [            8202]	unregistering all CEC clients
NOTICE:  [            8202]	unregistering client: libCEC version = 4.0.4, client version = 4.0.4, firmware version = 8, firmware build date: Fri Jul 21 09:08:24 2017 +0000, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0,  compiled on Thu Feb 21 12:07:56 UTC 2019 by jenkins@prometheus on Linux 4.15.0-43-generic (x86_64), features: P [... cut ...]
DEBUG:   [            8308]	unregistering all CEC clients
DEBUG:   [            9310]	communication thread ended
LibreELEC:~ #
```

```
LibreELEC (Milhouse): devel-20190501222748-#0501-g6b93ae6 (Generic.x86_64)
LibreELEC:~ # lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0a2b Intel Corp. 
Bus 001 Device 003: ID 2548:1002 Pulse-Eight CEC Adapter
Bus 001 Device 002: ID 20a0:0001 Clay Logic 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
LibreELEC:~ # 

```

SpaceAgeHero, 2019-05-13:
No longer interested.

closed 2019-05-13 by SpaceAgeHero