Pulse-Eight/libcec #346: Libcec 4, double inputs on remote

pelukessss, 2017-05-02:
Hi,
I own a RPi 3 with OSMC and, after upgrading to Kodi 17, every button I press on the remote is repeated twice. I reverted to Kodi 16.1 and the issue was gone, then I installed LibreELEC based on Kodi 17 and had the same problem. I tried the following settings with no luck

(These settings work on Jarvis and libcec 3xx)
Remote button press delay before repeating (ms): 300
Remote button press repeat rate (ms): 0
Remote button press release time (ms): 0

Remote button press delay before repeating (ms): 800
Remote button press repeat rate (ms): 0
Remote button press release time (ms): 0

Remote button press delay before repeating (ms): 300
Remote button press repeat rate (ms): 100
Remote button press release time (ms): 0

Remote button press delay before repeating (ms): 300
Remote button press repeat rate (ms): 100
Remote button press release time (ms): 100

Same issue seems to be present also in my RPi 2.

If I use Kore or Yatse to control Kodi there are no problems so I thought it might be a libcec issue.

My setup is Philips TV, Pioneer HP 071 Home Theater and Logitech Harmony 300 remote
(I had this setup also with Kodi 16.1)

Tell me if you need anything else to address the issue

Thanks

gdave321, 2017-06-03:
Appears to be caused by this change in libcec
==
From changelog libcec 3.1.0:
    skip double press removal. It is handled through other means. credits @popcornmix
==
After reverting that change here locally on libcec 4.0.2 my double/repeated keys are gone!

ChristianCiach, 2017-06-29:
Can confirm. Annoying double presses in Kodi 17 on Arch Linux using the USB adapter with a Sony TV (2017 model).

opdenkamp, 2017-07-12:
Can you try to stop Kodi, remove `~/.kodi/userdata/peripheral_data`, install libcec without the change reverted and then try Kodi again. I think that it may be caused by an old value of the settings.

Let me know if that doesn't fix it, and I'll reopen the ticket

closed 2017-07-12 by opdenkamp

opdenkamp, 2017-07-12:
Sorry, just seen this ticket. Looks like more people are having trouble. Keeping this closed as duplicate of https://github.com/Pulse-Eight/libcec/issues/326