Pulse-Eight/libcec #351: LibCEC 4.0.2 not working with Samsung LE32D550

thaleb, 2017-05-25:
I used to run Kodi on a Pi1 Raspbian with libcec 2.0.2 and never had any troubles, it was fully functioning. Now i've tried running Kodi on a Pi3 with LibreElec which had libcec 4.0.2 but it won't detect my TV. I've also tried different HDMI cables, but all the cables i've tried worked with the Pi1 libcec 2.0.2 combo. Whats wrong here?

Here's my cec-client output:

```
$ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [              61]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              62]     Open - vc_cec initialised
DEBUG:   [              62]     logical address changed to Free use (e)
NOTICE:  [              62]     connection opened
DEBUG:   [              63]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [              63]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              63]     << e0
DEBUG:   [              63]     processor thread started
DEBUG:   [            1064]     command 'feature abort' timeout
DEBUG:   [            1564]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [            1564]     << e0
DEBUG:   [            1598]     >> POLL not sent
DEBUG:   [            1598]     TV (0): device status changed into 'not present'
NOTICE:  [            1598]     registering new CEC client - v4.0.2
DEBUG:   [            1598]     detecting logical address for type 'recording device'
DEBUG:   [            1598]     trying logical address 'Recorder 1'
DEBUG:   [            1599]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            1599]     << 11
TRAFFIC: [            3144]     << 11
DEBUG:   [            4692]     >> POLL not sent
DEBUG:   [            4693]     using logical address 'Recorder 1'
DEBUG:   [            4694]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            4695]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            4695]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            4696]     Recorder 1 (1): CEC version 1.4
DEBUG:   [            4697]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
WARNING: [            4698]     RegisterLogicalAddress - vc_cec_set_logical_address(1) returned device is busy (3)
DEBUG:   [            4699]     UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [            4700]     logical address changed to Broadcast (f)
DEBUG:   [            4700]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            4700]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            4700]     GetPhysicalAddress - physical address = 4000
DEBUG:   [            4700]     AutodetectPhysicalAddress - autodetected physical address '4000'
DEBUG:   [            4700]     Recorder 1 (1): physical address changed from ffff to 4000
DEBUG:   [            4700]     << Recorder 1 (1) -> broadcast (F): physical address 4000
TRAFFIC: [            4700]     << 1f:84:40:00:01
DEBUG:   [            4700]     sending command 'report physical address' failed (4)
TRAFFIC: [            4701]     << 1f:84:40:00:01
DEBUG:   [            4701]     sending command 'report physical address' failed (4)
NOTICE:  [            4701]     CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 4.0.0.0, git revision: libcec-4.0.2+8-8563411~dirty, compiled on Sun Apr  2 17:04:01 UTC 2017 by root@hostname: Name or service not known on Linux 4.4.0-57-generic (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
waiting for input
DEBUG:   [            4701]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG:   [            4701]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [            4701]     << 10
DEBUG:   [            4701]     sending command 'feature abort' failed (4)
TRAFFIC: [            4701]     << 10
DEBUG:   [            4701]     sending command 'feature abort' failed (4)
DEBUG:   [            4701]     >> POLL not sent
DEBUG:   [            4702]     not sending command 'set osd name': destination device 'TV' marked as not present
DEBUG:   [            4702]     << requesting power status of 'TV' (0)
DEBUG:   [            4702]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [            4702]     << 10
DEBUG:   [            4702]     sending command 'feature abort' failed (4)
TRAFFIC: [            4702]     << 10
DEBUG:   [            4702]     sending command 'feature abort' failed (4)
DEBUG:   [            4702]     >> POLL not sent
DEBUG:   [            4702]     not sending command 'give device power status': destination device 'TV' marked as not present
```

kkkkamal, 2017-06-03:
I need to enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.

opdenkamp, 2017-07-12:
Your TV isn't responding to CEC polls. Please check if CEC has been enabled on your TV and if it's supported. If so, try another HDMI cable.

closed 2017-07-12 by opdenkamp

thaleb, 2017-07-18:
Did that already, if you read my comment you'd know. Thanks for nothing :+1:

opdenkamp, 2017-07-18:
I did read that, but it still looked like an issue with the HDMI cable if polls don't get a response. 
That error number (4 = interrupted system call) is not what you'd normally see when it's a problem with the cable though. It definitely doesn't show up when using our USB-CEC adapter. It could be an issue with the Pi's firmware (configuration) or perhaps something else is trying to use CEC at the same time as cec-client.

It's not a problem with libCEC or with our hardware, but something that's specific to the Pi, so I can't help you fix this. Try asking for help on the Rasberry Pi forums.

thaleb, 2017-07-18:
Thanks! This is an answer I can work with :)