Pulse-Eight/libcec #353: LibCEC 4.0.2 - Windows installator

Shamshala, 2017-05-27:
Hi,

after i updated the LibCEC to v4.0.2 i get an error "HRESULT: 0x8007007e" and then app crash everytime i try to launch it. If i've checked this problem on the internet properly, it has something to do with compilation process. Could you please fix that?

Thanks

yoGhurrt1, 2017-05-28:
Same here - Win 10 x64 Creators Update. I tried to reinstall everything few times, reinstalled some drivers (because it also had issue after update to CU), including installing LibCEC in location with lower security permissions like main folder on C: ,but nothing changed.

traufeisen, 2017-06-03:
Same here on Win 10 x64 Creators Update

Shamshala, 2017-06-03:
Ok, here it looks like deadvillage... Guys, install the **Microsoft Visual C++ 2015 Redistributable** (x86 and x64 to be sure), then new libCEC.

malard, 2017-06-03:
we're busy launching a new product so we havent had time to focus on open libCEC tickets. 

I assure you they will be looked into soon

closed 2017-06-05 by Shamshala