Pulse-Eight/libcec #355: gcc5 compilation error

trusnak, 2017-06-13:
[ 95%] Linking C executable cecc-client
cd /var/tmp/portage/dev-libs/libcec-4.0.2/work/libcec-4.0.2_build/src/cecc-client && /usr/bin/cmake -E cmake_link_script CMakeFiles/cecc-client.dir/link.txt --verbose=1
/usr/lib64/ccache/bin/x86_64-pc-linux-gnu-g++   -march=core-avx-i -O2 -fomit-frame-pointer -pipe -std=c++11  -Wl,-O1 -Wl,--as-needed CMakeFiles/cec-client.dir/cec-client.cpp.o CMakeFiles/cec-client.dir/curses/CursesControl.cpp.o  -o cec-client-4.0.2 -rdynamic -L/usr/lib64 -lp8-platform -lpthread -lpthread -ldl -lcurses -lrt
/usr/lib64/ccache/bin/x86_64-pc-linux-gnu-gcc  -march=core-avx-i -O2 -fomit-frame-pointer -pipe  -Wl,-O1 -Wl,--as-needed CMakeFiles/cecc-client.dir/cecc-client.c.o  -o cecc-client-4.0.2 -rdynamic -L/usr/lib64 -lp8-platform -lpthread -lpthread -ldl
cd /var/tmp/portage/dev-libs/libcec-4.0.2/work/libcec-4.0.2_build/src/cecc-client && /usr/bin/cmake -E cmake_symlink_executable cecc-client-4.0.2 cecc-client
make[2]: Leaving directory '/var/tmp/portage/dev-libs/libcec-4.0.2/work/libcec-4.0.2_build'
[ 95%] Built target cecc-client
CMakeFiles/cec-client.dir/cec-client.cpp.o: In function `PrintToStdOut(char const*, ...)':
cec-client.cpp:(.text+0x22f): undefined reference to `StringUtils::FormatV[abi:cxx11](char const*, __va_list_tag*)'
CMakeFiles/cec-client.dir/cec-client.cpp.o: In function `CecLogMessage(void*, CEC::cec_log_message const*)':
cec-client.cpp:(.text+0x4d7): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
CMakeFiles/cec-client.dir/cec-client.cpp.o: In function `ListDevices(CEC::ICECAdapter*)':
cec-client.cpp:(.text+0x1436): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
cec-client.cpp:(.text+0x15c3): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
cec-client.cpp:(.text+0x1605): undefined reference to `StringUtils::Left(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long)'
CMakeFiles/cec-client.dir/cec-client.cpp.o: In function `ProcessCommandSELF(CEC::ICECAdapter*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
cec-client.cpp:(.text+0x3bb6): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
CMakeFiles/cec-client.dir/cec-client.cpp.o: In function `ProcessCommandLANG(CEC::ICECAdapter*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
cec-client.cpp:(.text+0x52f3): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
CMakeFiles/cec-client.dir/cec-client.cpp.o: In function `ProcessCommandSCAN(CEC::ICECAdapter*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
cec-client.cpp:(.text+0x5e9f): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
cec-client.cpp:(.text+0x5f08): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
cec-client.cpp:(.text+0x5f55): undefined reference to `StringUtils::Format[abi:cxx11](char const*, ...)'
CMakeFiles/cec-client.dir/cec-client.cpp.o:cec-client.cpp:(.text+0x5fae): more undefined references to `StringUtils::Format[abi:cxx11](char const*, ...)' follow
collect2: error: ld returned 1 exit status
make[2]: *** [src/cec-client/CMakeFiles/cec-client.dir/build.make:121: src/cec-client/cec-client-4.0.2] Error 1

# gcc --version
gcc (Gentoo 5.4.0-r3 p1.3, pie-0.6.5) 5.4.0

opdenkamp, 2017-07-12:
It looks like it's simply missing the platform library on your system. See https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md

trusnak, 2017-07-12:
Thank you. Reemerge platform library solved the problem:
# emerge -1av dev-libs/libplatform

closed 2017-07-12 by trusnak

opdenkamp, 2017-07-12:
I'll email the maintainer of the libCEC package on Gentoo that this dependency is missing. Thanks for reporting back.