Pulse-Eight/libcec #356: Import fails with Python 3.5/3.6 and (possible) fix

smithfred, 2017-06-20:
With Python 3.5/3,6 I get the following (just the Python version number and line numbers below vary):

> $ python3
> >>> import cec
> Traceback (most recent call last):
>   File "/usr/lib64/python3.5/site-packages/cec/__init__.py", line 18, in swig_import_helper
>     return importlib.import_module(mname)
>   File "/usr/lib64/python3.5/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 986, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 956, in _find_and_load_unlocked
> ImportError: No module named '_cec'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/usr/lib64/python3.5/site-packages/cec/__init__.py", line 21, in <module>
>     _cec = swig_import_helper()
>   File "/usr/lib64/python3.5/site-packages/cec/__init__.py", line 20, in swig_import_helper
>     return importlib.import_module('_cec')
>   File "/usr/lib64/python3.5/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> ImportError: No module named '_cec'
> >>>

There seem to be similar problems in a few bugs (#316, #329 and possibly a few others).

#316 was closed with:

> it seems like swig/swig#769 is causing your problems
and
> Because this is not a problem in libCEC or the way we use Swig or Python

However, a comment on swig/swig#769 mentions:

> we used to rename the generated .py file to __init__.py and this stopped working recently. The SWIG docs seem to suggest that this isn't expected to work, so we assumed it was just by chance that it did before the recent changes

An example layout in SWIG's docs (http://www.swig.org/Doc3.0/Python.html#Python_modulepackage) seem to confirm this:

> then the resulting directory layout should be

> pkg1/
> pkg1/__init__.py
> pkg1/pkg2/__init__.py
> pkg1/pkg2/foo.py        # (generated by SWIG)
> pkg1/pkg2/_foo.so       # (shared library built from C/C++ code generated by SWIG)

However, libcec seems to be doing the same as the commenter on the SWIG bug was, namely renaming cec.py to __init__.py (from CheckPlatformSupport.cmake):

>      install(FILES       ${CMAKE_BINARY_DIR}/src/libcec/cec.py
>              DESTINATION lib/python${PYTHON_VERSION}/${PYTHON_PKG_DIR}/cec
>              RENAME      __init__.py)

Which seems not to be the correct approach.

Sure enough, renaming cec/__init__.py back to cec/cec.py fixes the problem and 'import cec' works again.

An __init__.py isn't required for Python 3.3+ (because of PEP 420). I think adding an (empty) __init__.py should be sufficient for earlier Python versions?

opdenkamp, 2017-07-12:
c+p from another ticket about this: `I'll have to check updating Python in the next major release. Too much risk of breaking existing things when doing it in a minor release.`

opdenkamp, 2017-07-18:
https://github.com/Pulse-Eight/libcec/commit/5b4faecdeb1d6f8b27822fc50f4b49f6f5c66b95 should stick older versions in the /cec/ subdir, and newer one level up. I've checked this with the pyCecClient testing app on Debian and Win10 with Python 2.7. Please let me know if this fixes the problem, then I can close another ticket.

smithfred, 2017-11-18:
Any idea when there might be a major release to incorporate this, since libcec is basically broken on the current Python release? Thanks :)

lulandco, 2017-12-18:
Is there a workaround to import module in python3 on ubuntu artful ? I've read a lot thinks without finding a way to figure out.

MilhouseVH, 2018-11-05:
LibreELEC builds now fail after this commit: http://ix.io/1r4A

opdenkamp, 2018-11-06:
I've fixed the error at the end and the warning about the snprintf() call (will be pushed in a few), but I can't reproduce the swig warnings in the generated code.

I've set up a debian sid install with the same environment as *elec: gcc 8.2.0, python 2.7.15, swig 3.0.12. It builds the deb without warnings.

Perhaps it's some custom patch/config used by the build system or some cross compile issue, but since *elec doesn't use libCEC's python wrapper afaik, you can disable python support when building.

MilhouseVH, 2018-11-06:
The build is successful when adding `-DSKIP_PYTHON_WRAPPER=1` - thanks.

Hopefully nobody is using/relying on the `cec` import...

opdenkamp, 2018-11-06:
Kodi has a direct integration and doesn't use the python wrapper. It's a very odd combination to use anyway (bleeding edge gcc with python 2.7), which I doubt has ever worked on *elec because of the warnings which aren't caused by the changes linked here. Not sure where those are coming from and I've spent too much time on this already ;) You can try to use a more recent (3.x) version of python instead.

MilhouseVH, 2018-11-06:
When building LibreELEC 9 with Kodi we still deploy Python 2.7.15 on the target, as that is what Kodi is still using. We'll upgrade to Python 3.x on the target once Kodi makes the move to Python 3.x.

A LibreELEC 9 build that includes the wrapper doesn't show any errors, so it appears to be working:
```
LibreELEC (Milhouse): devel-20181105210235-#1105-g4585899 (RPi2.arm)
rpi22:~ # python
Python 2.7.15 (default, Nov  4 2018, 14:17:59)
[GCC 8.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cec
>>> c=cec.libcec_configuration()
>>> print(c)
<cec.libcec_configuration; proxy of <Swig Object of type 'CEC::libcec_configuration *' at 0x76aeeb90> >
>>> cec.CEC_LOG_TRAFFIC
8
>>> quit()
```

Dropping the wrapper is no big deal and a quick fix - like you, I don't want to waste any more time on this than I have to. :)

opdenkamp, 2018-11-06:
Yeah but it may fail when detecting adapters, because it was complaining about a missing copy-assignment in one of the structs/classes that it uses in the code generated by swig. Not sure why it wants that when doing a memset, but I suppose that (this version of) c++ does more than just changing memory.

closed 2018-11-06 by opdenkamp

MilhouseVH, 2018-11-06:
I've just built `libcec` with db58911b8002d6f2694babd9b322fcfd958cdf72, and now I'm able to build `libcec` both with and without the wrapper, so that's all good - thanks.

I'll drop the the wrapper from the next bump as it's quite large (400KB) and if it's not being used (and may not even work) then that's 400KB we don't need.