Pulse-Eight/libcec #357: EventGhost

kdschlosser, 2017-06-22:
I am one of the developers for EG and we have some users that are wanting to use libCEC along with the supplied plugin and have not been able to.

There is also a new version of EG that is close to being a full release. Python has been update to 2.7

I believe there is in issue (at least when running on EG 0.5) where the _cec.pyd file has been compiled using python 2.6 and will not run properly with python 2.7. I am not 100% sure on this but here is the traceback

    Error loading plugin file: C:\Program Files (x86)\EventGhost\plugins\libCEC
    Traceback (most recent call last) (WIP-2017.06.14-17.35.39):
      File "C:\Program Files (x86)\EventGhost\eg\Classes\PluginModuleInfo.py", line 64, in __init__
     __import__(moduleName, None, None, [''])
    File "C:\Program Files (x86)\EventGhost\plugins\libCEC\__init__.py", line 45, in <module>
      import cec
    File "C:\Program Files (x86)\EventGhost\plugins\libCEC\cec\__init__.py", line 28, in <module>
    _cec = swig_import_helper()
    File "C:\Program Files (x86)\EventGhost\plugins\libCEC\cec\__init__.py", line 24, in swig_import_helper
    _mod = imp.load_module('_cec', fp, pathname, description)
    ImportError: DLL load failed: The specified module could not be found.

and if i add a python26.dll it spits out an error about using the python26.dll file when running python 2.7 instead of the previous traceback 

Not sure how to handle this. You can drop me a line on here or in a message on the EventGhost forum. either is fine. 

Thanks again for you assistance

opdenkamp, 2017-07-12:
c+p from another ticket about this: `I'll have to check updating Python in the next major release. Too much risk of breaking existing things when doing it in a minor release.`

I'll look into it after the minor release with fixes is out, thanks.

topic2k, 2017-07-12:
Is it possible that we build/compile it ourself?

opdenkamp, 2017-07-12:
Apparently there is a problem with Swig+Python2.7+libCEC which prevents it from working. It could simply be this https://github.com/Pulse-Eight/libcec/issues/329 which you can try.

topic2k, 2017-07-12:
Thanks for the quick answer. As i have only little expirence with Python, what are the commands to get the _cec.pyd file?

opdenkamp, 2017-07-12:
Unfortunately that isn't easy if you're not familiar with building things from source on Windows. Instructions can be found here, but they assume that you know how most of this works: https://github.com/Pulse-Eight/libcec/blob/master/docs/README.windows.md

opdenkamp, 2017-07-18:
@kdschlosser I've just checked the build with Python 2.7 on Windows. It does work, but `_cec.pyd` needs to be one level up (so not in a /cec subdir, at least on Windows with this Python version), and cec.dll has to be in the same directory or in the system search path.

I've just checked, and your error message is what shows up when it can't find cec.dll. The wrong path for the pyd file results in `ImportError: No module named _cec`

kdschlosser, 2017-07-18:
Oh sweet. Thanks for the info.

kdschlosser, 2017-07-19:
OK I tried this and it does not work. I tried it in EventGhost. and I also tried it directly in a python installation..

Maybe I am misunderstanding.

In EventGhost the folder where the plugin resides is libcec
and _cec.pyd also cec.dll if in libcec/cec
so what i did was I moved those 2 files to libcec

still get the same error.

Then I tried in a Python installation. i copied the cec from the plugin into site-packages and tried to do an import. didn't work..

then I tried moving those 2 files into the site_packages folder. still didn't work.
maybe i am doing something wrong. 

I get the same error from all of the above

opdenkamp, 2017-07-19:
`_cec.pyd` should be one level up. If you build the master branch of libCEC now (use `/windows/create-installer.cmd`) then it'll be put in the correct directory.

I've just done another test build (untested, not signed) which may be helpful: https://www.dropbox.com/s/cq5vt5ahdvls7fr/libCEC-4.0.2.exe?dl=1

kdschlosser, 2017-07-20:
OK cool I got ya ..

I didn't realize we were talking that it had to be built. I missed that. No error now which is a good thing.

I do have a couple of other questions. Me and the other guy that chimed in there @topic2k, we are attempting to give new life to EventGhost. it has needed to be upgraded in a very long time. and we have done this. and along the way we have also made some changed to thing that are going to be a great benefit. One of the things that was added to EG before our time was a .egplugin file. Tho it was not complete. We have completed this feature and it will allow for the installation of the plugin by EG and it would be put into a user plugin directory as well as re scan the plugin list so there would be no need to restart EG. I do not know how your installer installs the files. but if it is able to call EventGhost with the location of the egplugin file EG can take over from there. 

also the plugin only allows support for a single cec adapter. Could I update the plugin to allow for multiple adapters and submit a PR? or do you want to do it? 

I do not know if you want me to make separate issue for these or not. let me know if this is something that you want to do or if we should. We are going to be rolling out a full release of EG soon along with moving the website to a VPS and giving the site a facelift at the same time and the cec adapter working the way the users requested (and we have quite a few of them) would be a nice addition to the "grand re opening" 

Thank You for all of your help it is going to make people very happy this is now working again.

opdenkamp, 2017-07-20:
We use NSIS as installer, which can call an external program while installing afaik. This is done by searching for the installation path in the registry now, and simply copying the required files into the installation directory: https://github.com/Pulse-Eight/libcec/blob/master/project/libCEC.nsi#L293

As for multiple adapters, this version of libCEC only supports one adapter. If you want to talk to multipe adapters, then you'll have to create an instance of the addon for each adapter (so each instantiates it's own libCEC), and tell libCEC which com port to talk to.

I haven't looked at EventGhost in detail, so I don't know what's needed to hook up `.egplugin`. Patches to support this would be very welcome.

kdschlosser, 2017-07-20:
the egplugin file is very simple. it is a zip file that has the plugin (with it's directory) and a file info.py. at the root of the zip. in the info .py file are the keyword arguments for eg.RegisterPlugin as attributes with the values set to what is in the plugins eg.RegisterPlugin. and the extension is renamed. 

to use the egplugin file all that would have to be done is you would have the installer run eventghost with the file path\file name after it. 

EG will do the rest.

and EG does support multiloading a plugin. this has to be set in the eg.RegisterPlugin

    canMultiLoad=True

but it would appear as tho the way the plugin is currently there is no concession for selecting a specific adapter it will always grab the last one in the adapters that are returned. I do not have one of these adapters. so i am unable to test things directly.

 I was thinking more long the lines of having the plugin poll the adapters for the devices they have attached to them and list the devices. so the user would be able to select which adapter to use based on that. and simply pass the strComName to the plugin when it starts

I am also trying to expand the plugin to encompass all of the features that libcec has to offer.
an example would be discrete functions for things like Mute On and Mute Off. and to have a list of specific buttons that can be pressed and add these as actions as well. 

I am going to poke about the code a bit and see if I can figure out how this works. and If you do not mind, what I can do is submit a PR for the plugin portion and if it is good and you want to merge it then great or if you want to leave it as is. that is ok as well. It just seems that the plugin is really only scraping the surface of what is available. and I would much rather have the plugin distributed with the original package then to have a spinoff of it. someone has already tried to do a spinoff that made us of cec-client.exe.  I do not know how well it worked. but using subprocess and grabbing output is not the best way to go.

I also wanted to note then when i run the tray i get the message no adapters are found. and when i click on No to scan again the tray crashes. (Windows 7 sp1 x64)

topic2k, 2017-07-20:
A few days ago a user with two adapters wrote about the problem with multiload. I made some changes to the plugin and it seems to be working (no bug report until). [Link to forum thread](http://www.eventghost.net/forum/viewtopic.php?f=2&t=9742&hilit=pulse+eight)

Here is the changed plugin `__ini__.py`: [EventGhost-plugin.zip](https://github.com/Pulse-Eight/libcec/files/1163894/EventGhost-plugin.zip)

kdschlosser, 2017-07-21:
@topic2k 
yeah the reason why it wouldn't work was because it would always select the last adapter in the list of adapters. I am going to take a look at your modifications and see if we can't get some insight into how to set this up in the best possible manner so the user can get the most from libCEC.

I also sent in to libCEC the needed documents for contributing so that is already taken care of. what I can do is fork the repo. and we can make any adjustments to in in a joint manner. and then I can submit a PR off of my fork.

kdschlosser, 2017-07-21:
@topic2k 

This is funny. I am looking at the changes you made and I have done all of the same things. almost verbatim. 

I see now how to use the device com port as well. I had just started to read the cec\__init__.py

opdenkamp, 2017-07-21:
@kdschlosser fyi: received the agreement, thanks.

passing the port is a setting in the configuration struct, and you can find out which adapters are detected by the system with the DetectAdapters call

kdschlosser, 2017-07-21:
I do have to say I am impressed with the amount of control/information that libCEC can do. its very impressive. great work. this is a means to bypass things like the Samsung IP control being locked out. if something only existed to interface with my clothes washer and dryer.... LOL they are "smart". but only work with Samsung app which is "dumb".

 I hope the agreement is in order. I wanted to get ahead of the curve and get that taken care of. 

I have thought of a way to make it so the plugin would not have to be multiloaded. 

I do have a question tho. there is a program called VirtualHere and what it is used for is it will do USB port redirection on a raspberry Pi to a PC. My thought on this would be the ability to control a TV without having to have a computer near the TV. use the Raspberry Pi to send the USB connection over to a single "Server". Do you know if the CEC adapter would work being redirected? I use this kind of a setup for external DVD drives, video game controllers and a memory card reader in my home. And it works without flaw. I have even been able to flash the firmware on micro controllers over it as well (which is a com port connection). 

I have put together the list of button presses and have them set up to be added as actions. I will be taking a day off from working on my house in the next few days. and I will hammer out most of the plugin then. Do you have a API write up on this? something that specifically pertains to the python lib?

kdschlosser, 2017-07-21:
I do also want to state that I am by no means a seasoned programmer. If I ask what would seemingly be a simple thing it's because I'm new and still learning. I have about 2 1/2 years from the day I typed my first line of code. So I want to apologize ahead of time for my lack of knowledge.

Another question I have is when do you think your next major version change will be? or do you have an idea as to when?. 
Also is there any kind of a developers "blackbox" something that would mimic a CEC adapter that is software based?. or should i order an adapter?

kdschlosser, 2017-08-12:
OK so I had a user of EventGhost donate an adapter to me. I have been working on the plugin and I would like to know if there is a way I am able to query for the current volume level?

kdschlosser, 2017-08-12:
OK here is what I am going to do. since there is no doc strings for the cec.py file. I am going to key them in from what I know. If someone could help fill in the blanks that would greatly be appreciated. I will create a PR for the ones that I do.

malard, 2017-08-12:
if you want to know the raw cec commands, you can find them in the HDMI documentation or at www.cec-o-matic.com

kdschlosser, 2017-08-15:
@malard

I didn't want to know the raw commands. But i did learn something from cec-omatic so TY for posting that.

The AudioStatus returns both the volume and the mute state at the same time. I did not realize this I had to do some bitwise operations on the returned value in order to get the current volume level, One thing I am unsure of is the docs state that the colume supported is between 0 and 127 i believe. does that mean if the volume is at 100% on the device it is going to return a 127? I tried the cec_tray and the volume buttons are grayed out not sure why they are my projector does have a speaker in it.

also don't know if you wanted to add this to the list of compatible manufacturers, Epson projectors do work with this adapter. I have not gone through and tested all of the specific functions yet. I have a PowerLite Home Cinema 3020.

I put together the start of a new plugin. I posted it on the forums for EventGhost. and I have also attached it here if anyone wants to give it a go. I am sure it has issues. I changed the folder, guid, class name for the plugin as not to cause issues if the libCEC plugin is in the plugins directory. The plugin cannot be running at the same time as the one included in the current or past releases of libCEC so a new tree would need to be made first.

extract the file that is inside of the zip and double click it to have EventGhost install the plugin. EventGhost will put the plugin into %programdata%\EventGhost\plugins

[PulseEight-1.0b.zip](https://github.com/Pulse-Eight/libcec/files/1224220/PulseEight-1.0b.zip)

malard, 2017-08-15:
did you actually get a reply from the projector when you requested the AudioStatus? if not, that will be why it is greyed out. (i think)

kdschlosser, 2017-08-15:
yes I have gotten reply's from the projector for an audio status. But I am thinking that it might be because there was no video stream going to the projector. Because I was not able to adjust the volume using this plugin either. right now it's hard for me to test anything out because I have most of my home automation and media room torn apart due to construction on my home. and I also have all of my HTPC's out of service due to making a single chassis for them. space problems. needed to fit 4 full ATX main boards with double slot full length video cards the power supplies and the rest of the stuff to go along with it all in a space of 19" wide by 14" high. The CEC sparked my interest because the serial on my projector is non functioning.

kdschlosser, 2017-08-15:
Here is a more updated version the plugin. I corrected a bunch of stuff. I also added events for changes to volume, mute, source, device connection, menu language,  power.

[PulseEight-1.1b.zip](https://github.com/Pulse-Eight/libcec/files/1225623/PulseEight-1.1b.zip)

seanwalter, 2017-10-08:
Kdschlosser, the 1.1b plugin is really nice. Thank you for your efforts in pulling it together. I am noticing it doesn't seem to detect the events when the TV is powered up or down via the remote, though Opdenkamp's older version does this.

I am not exactly sure what would need to change in 1.1b to bring this back, but thought I'd point this out in case there is a way to include that. If there is a better way to capture whether the remote is powering the device up or down, I'm open to adjusting. Again, thank you!

kdschlosser, 2017-10-08:
@seanwalter 

This is a beta of the plugin I am still testing it and working through some issues. If you would like to follow and help with the beta testing that would be great. you can get the latest version here
http://www.eventghost.net/forum/viewtopic.php?p=48647#p48647

and that same forum thread is where any information is and where i will be putting updates to the plugin until it is finished up. then i will update the one here so it can be added

kdschlosser, 2018-02-15:
I started working on this plugin again There are some thing that I would love to have some clarification on and also see if there is a way to solve some issues I am having.

first thing.

VolumeUp, VolumeDown, AudioMute, AudioUnmute, AudioToggleMute, AudioStatus
do the above only work if there is an AVR attached?

If that is the case is there a way to do the same if there is not?

second. 
SetMenuStateCallback, SetSourceActivatedCallback, SetCommandCallback, SetKeyPressCallback seem to not function. This could be something I am doing wrong. Not sure what it is. could I get some information on use?
what does EnableCallbacks do? and how to use it

third
AudioStatusToString returns TODO?
MenuStateToString there is no GetMenuState function/method there is a callback SetMenuStateCallback but I would like to know if is this does work?
OpcodeToString, I do not know what the purpose of this is, information would be most helpful.

fourth
other functions/methods that I would like to have some information on please.
SwitchMonitoring
PollDevice
SetStreamPath
InitVideoStandalone
UserControlCodeToString

and last.
There seems to be incorrect reporting of a devices power state. Well not sure if it is an incorrect or not. Let me explain.

I have set up EventGhost to poll the adapter for device connection, active device, device power, audio mute, audio volume, menu language, source. This is so that I can cause Events to take place inside of EG so the user is able to perform actions based on these events. So say you want to have the lights dim in your media room when the TV gets turned on. This would be able to be achieved through the use of these events. I used the Power of a device specifically because in my case I only have a TV connected and it will report that the TV has powered on or off when it has not changed state. I have read somewhere that just because you ask the adapter to do something it doesn't mean that it will get an answer on the network. How does libCEC handle this issue? The other thing that I have read is that with CEC the order in which a device receives commands may not be the order in which replies are sent. Since I am running a polling loop that is done in a separate thread and other commands can get sent at the same time. I thought this may have been the issue so I have put into place a thread locks but the problem still persists. 

I look forward to the information/help 

Thanks again
K

kdschlosser, 2018-02-15:
One other thing..

Is there some kind of a supported feature list for specific brands/models?

I have tested the adapter on 2 different devices in regards to sending remote key presses and the only one that seems to work is the power off on both. I have used cec-o-matic and also tried sending raw codes from that but nothing happens.