Pulse-Eight/libcec #359: Firmware upgrade under Linux

Anonym-tsk, 2017-07-10:
How to upgrade firmware of USB-cec?
cec-firmware-upgrade no longer supported and was removed from ppa.

opdenkamp, 2017-07-10:
You'll have to use Windows to upgrade the firmware. We can't support Linux firmware upgrades at the moment.

Anonym-tsk, 2017-07-10:
Maybe there is some way? Manually flash the new version using the old cec-firmware-upgrade?

opdenkamp, 2017-07-10:
Which device do you have and which firmware version is it running now?

Anonym-tsk, 2017-07-10:
https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter
Firmware v4

opdenkamp, 2017-07-10:
That firmware version should be fine then for this device. Any reason why you want to upgrade the firmware?

Anonym-tsk, 2017-07-10:
No reason, just new version. Do you have changelog?

opdenkamp, 2017-07-10:
No, there's no changelog available unfortunately. Unless you're having trouble, you shouldn't have to upgrade the firmware.

closed 2017-07-10 by opdenkamp