Pulse-Eight/libcec #361: Sharp Aquos Link TV - Some keys that should work produce strange errors and are not working

Sequajo, 2017-07-16:
Some keys on the Sharp Aquos Link remote are not working, but are sending a signal to libcec.

Best example are the red,green,yellow and blue keys, they are not working and i got errors like this in kodi log.

DEBUG: PushCecKeypress - received key a7 duration 0 (rep:0 size:0)
DEBUG: PushCecKeypress - added key a7
DEBUG: PushCecKeypress - received key a7 duration 207 (rep:0 size:0)
DEBUG: PushCecKeypress - ignored key a7

DEBUG: PushCecKeypress - received key a6 duration 0 (rep:0 size:0)
DEBUG: PushCecKeypress - added key a6
DEBUG: PushCecKeypress - received key a6 duration 210 (rep:0 size:0)
DEBUG: PushCecKeypress - ignored key a6

Should these keys work, or is the TV blocking these keys?

opdenkamp, 2017-07-18:
`PushCecKeypress` is a method within Kodi, which indicates that libCEC is working correctly and forwarding the keys to Kodi. If it's not doing anything, then it's either a bug in Kodi, or the key mapping isn't set up correctly.

You can find more info about this here: http://kodi.wiki/view/CEC

Sequajo, 2017-07-18:
THX for your answer.
I already tried the general keymap file for remote.xml.
With these remote.xml i got the above kodi.log.
So i think some of the buttons on my remote are still unknown for libCEC.
Is there a way for me to activate them, because you closed the topic.
I thought this information for the buttons is in the libCEC Code and for some of my remote keys it is missing.

opdenkamp, 2017-07-18:
Ok, I see what's going on here now. Looks like Sharp invented some new keycodes, `0xA6` and `0xA7` aren't valid according to the last version of the specification that I've seen.

Could you stop Kodi, start cec-client and type `as` + enter.
Then press the red, green, yellow and blue key (in this order).
Exit the app by typing `q` + enter.

Then send me the full output of cec-client.

I closed the ticket earlier because `PushCecKeypress` was being called, which means that keys are actually being forwarded to Kodi, but it won't recognise the codes without mapping them manually.

Sequajo, 2017-07-18:
I tried that, i used ssh to connect to console ( i havent stopped kodi i am using libreelec system )
cec-client
as + enter
( then all keys that are already working got logged, but the ones that are not working there was nothing logged at all here )
q + enter

So as result there is no useable log with the not working keys which are recognized by PushCecKeypress :(

Cygn, 2021-04-29:
Hey there, sorry for necro-posting - but I've got a very similar issue with a ONKYO avr remote, with a LibreELEC 9.2.6 on a rpi-2B. 

None of the keys has any effect on kodi, although they appear in the kodi.log as PushCecKeypress events. See my log [here](http://ix.io/3ljr).

I've also tested [cec-client with just left/right/down buttons](http://ix.io/3ljs):
The strange thing, is cec-client seems to label them correctly as up/down... but kodi sees a7, a8, and does not generate any OnKey actions.

Any pointers will be highly appreciated! Thx in advance