Pulse-Eight/libcec #362: Installation issue on Raspberry Pi

vinjex, 2017-07-23:
I've followed the raspberry installation instruction carefully on a fresh install. Everything works fine except when trying to:
`cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..`
So I can't continue the installation.
I'm using hassbian if that matters.
this is the error I get:
```
CMake Error at src/libcec/cmake/CheckPlatformSupport.cmake:193 (if):
  if given arguments:

    "3" "GREATER" "2" "AND" "3" "GREATER_EQUAL" "7"

  Unknown arguments specified
Call Stack (most recent call first):
  src/libcec/CMakeLists.txt:133 (include)
```
and the logs:
CMakeError.log:
```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pi/libcec/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_d0101/fast"
/usr/bin/make -f CMakeFiles/cmTC_d0101.dir/build.make CMakeFiles/cmTC_d0101.dir/build
make[1]: Entering directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_d0101.dir/CheckSymbolExists.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_d0101.dir/CheckSymbolExists.c.o   -c /home/pi/libcec/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_d0101
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d0101.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_d0101.dir/CheckSymbolExists.c.o  -o cmTC_d0101 -rdynamic 
CMakeFiles/cmTC_d0101.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x38): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_d0101.dir/build.make:97: recipe for target 'cmTC_d0101' failed
make[1]: *** [cmTC_d0101] Error 1
make[1]: Leaving directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_d0101/fast' failed
make: *** [cmTC_d0101/fast] Error 2

File /home/pi/libcec/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/libcec/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e419c/fast"
/usr/bin/make -f CMakeFiles/cmTC_e419c.dir/build.make CMakeFiles/cmTC_e419c.dir/build
make[1]: Entering directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_e419c.dir/CheckFunctionExists.c.o
/usr/bin/cc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_e419c.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.6/Modules/CheckFunctionExists.c
Linking C executable cmTC_e419c
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e419c.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_e419c.dir/CheckFunctionExists.c.o  -o cmTC_e419c -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_e419c.dir/build.make:97: recipe for target 'cmTC_e419c' failed
make[1]: *** [cmTC_e419c] Error 1
make[1]: Leaving directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_e419c/fast' failed
make: *** [cmTC_e419c/fast] Error 2

Determining if the function initscr exists in the curses failed with the following output:
Change Dir: /home/pi/libcec/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_8b05e/fast"
/usr/bin/make -f CMakeFiles/cmTC_8b05e.dir/build.make CMakeFiles/cmTC_8b05e.dir/build
make[1]: Entering directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_8b05e.dir/CheckFunctionExists.c.o
/usr/bin/cc    -DCHECK_FUNCTION_EXISTS=initscr   -o CMakeFiles/cmTC_8b05e.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.6/Modules/CheckFunctionExists.c
Linking C executable cmTC_8b05e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8b05e.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=initscr    CMakeFiles/cmTC_8b05e.dir/CheckFunctionExists.c.o  -o cmTC_8b05e -rdynamic -lcurses 
/usr/bin/ld: cannot find -lcurses
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_8b05e.dir/build.make:97: recipe for target 'cmTC_8b05e' failed
make[1]: *** [cmTC_8b05e] Error 1
make[1]: Leaving directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_8b05e/fast' failed
make: *** [cmTC_8b05e/fast] Error 2

Determining if files tda998x_ioctl.h;comps/tmdlHdmiCEC/inc/tmdlHdmiCEC_Types.h exist failed with the following output:
Change Dir: /home/pi/libcec/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_4d58a/fast"
/usr/bin/make -f CMakeFiles/cmTC_4d58a.dir/build.make CMakeFiles/cmTC_4d58a.dir/build
make[1]: Entering directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_4d58a.dir/CheckIncludeFiles.c.o
/usr/bin/cc    -fPIE   -o CMakeFiles/cmTC_4d58a.dir/CheckIncludeFiles.c.o   -c /home/pi/libcec/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c
/home/pi/libcec/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:27: fatal error: tda998x_ioctl.h: No such file or directory
 #include <tda998x_ioctl.h>
                           ^
compilation terminated.
CMakeFiles/cmTC_4d58a.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_4d58a.dir/CheckIncludeFiles.c.o' failed
make[1]: *** [CMakeFiles/cmTC_4d58a.dir/CheckIncludeFiles.c.o] Error 1
make[1]: Leaving directory '/home/pi/libcec/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_4d58a/fast' failed
make: *** [cmTC_4d58a/fast] Error 2

Source:
/* */
#include <tda998x_ioctl.h>
#include <comps/tmdlHdmiCEC/inc/tmdlHdmiCEC_Types.h>

int main(void){return 0;}

```
If needed I can add CMakeOutput.log.

john-ruddy, 2017-07-23:
I just got the exact same error.
Raspberry Pi 3, not a fresh install but it's been running Home Assistant (All-In-One installer) for months with no issue

john-ruddy, 2017-07-23:
There was a new commit to that file on July 18
https://github.com/Pulse-Eight/libcec/commit/5b4faecdeb1d6f8b27822fc50f4b49f6f5c66b95

vinjex, 2017-07-23:
Thank you so much. Maybe this will help others in this situation.

john-ruddy, 2017-07-23:
Don't close the issue, there's definitely an error in the code but I don't know this syntax to be able to contribute a real fix. The project owners need to address this.

opdenkamp, 2017-08-01:
It's a bug that got introduced 2 weeks ago indeed, in https://github.com/Pulse-Eight/libcec/commit/5b4faecdeb1d6f8b27822fc50f4b49f6f5c66b95
I'll fix it up

dmikhalsky, 2017-08-16:
https://github.com/home-assistant/hassbian-scripts/issues/38

the workaround worked for me

stevenazari, 2017-09-20:
Hi there,
is there an update on this?

ajmas, 2017-10-10:
Just got the latest HEAD and this still appears to be an issue.

PR #409 (2018-11-02): Fix broken Python version check and failure to build on cmake < 3.7
Fixes #362.  Also affects #356.

Ubuntu 16.04 and derivative distributions ship cmake 3.5, which does
not support GREATER_EQUAL.  Tested fix on Ubuntu MATE 16.04 for
Raspberry Pi.

Also, the existing logic to check for Python 2.7+ does not work.
This patch corrects that, and I was able to 'import cec' in
Python 3 on Ubuntu MATE 16.04 for Raspberry Pi without error.
closed 2018-11-02 by opdenkamp