Pulse-Eight/libcec #363: [question] LG TV is not responding to standby or on CEC command

sodd, 2017-07-24:
Hi,
first of all i would like to thank you for yours awesome work!

So my problem is in putting my LG tv (43lh510v) to **standby** or **on** power status from odroid c2 (libCEC version 4.0.2). I'm able to control KODI by remote controller but when I send in cec-client **standby 0** or **on 0** command nothing happens. I thought I can only read cec communication from TV but when i send **pow 0** command I actually receive current TV status so I my conclusion is I'm able to send some command to my tv.

I'm able to turn on TV by restart kodi service witch I think is done by "Auto Power Sync" feature. But still i cant to turn off the TV.

What I'm trying to achieve is turn the tv off when kodi is going to idle mode and power tv on when want to use kodi.

Right now I'm stuck and don't know what to do to achieve my goal.

Thank you for any advice.

malard, 2017-07-24:
it is known that it is not possible to set Standby on an LG TV, unless they have resolved that issue in the most recent range of TVs this wont be possible

sodd, 2017-07-24:
thx for reply
one think i don't understand is the Auto Power Sync. If this feature is there to synchronize TV Standby with another device there must by a way to use it for my purpose or am I wrong?

malard, 2017-07-24:
LG do not support Power Off (Standby) in their CEC implementation

sodd, 2017-07-25:
Okay, just for the record what I saw the last night. 
If the KODI goes to sleep mode, the TV will automatically switch to Standby after a while. I wasn't too patient to realize the "Auto Power Sync" feature actualy works.

closed 2017-07-25 by sodd

filimonic, 2019-05-26:
Just FYI. Some new LG TVs support standby

filimonic, 2023-01-12:
Hardware workaround for LG TVs with WebOS not supporting CEC STANDBY https://github.com/ALFISYS/cec_lg_off