Pulse-Eight/libcec #364: TV powers back off after switching off

popcornmix, 2017-07-30:
Using a Milhouse nightly LE build the TV will power back on after powering it off.
This is unwanted.

Powering off the AVR before the TV seems to avoid this.
This wasn't required prior to the recent updates (approx 2 weeks ago).

Here is CEC enabled log:
http://sprunge.us/PHWI

popcornmix, 2017-07-30:
Seems to be caused by the "when tv is powered off" action.
Stop causes TV to power on.
Pause/ignore doesn't.
And I think it is screensaver related.

I suspect dim screensaver is stopped by the stop command and that turns TV back on.

opdenkamp, 2017-08-28:
```
16:14:13.734 T:1942373280   DEBUG: CAnnouncementManager - Announcement: OnScreensaverActivated from xbmc
16:14:13.734 T:1942373280   DEBUG: GOT ANNOUNCEMENT, type: 4, from xbmc, message OnScreensaverActivated
16:14:13.739 T:1942376464   DEBUG: CAddonSettings[screensaver.xbmc.builtin.dim]: loading setting definitions
16:14:13.739 T:1942376464   DEBUG: CAddonSettings[screensaver.xbmc.builtin.dim]: trying to load setting definitions from old format...
16:14:13.739 T:1942376464   DEBUG: ------ Window Init () ------
```
screensaver is activated, but the setting to power off the tv isn't enabled in Kodi (or someone broke something ;))

```
16:14:53.613 T:1942376464  NOTICE: NFS is idle. Closing the remaining connections.
16:15:02.163 T:1574278048   DEBUG: CecLogMessage - >> 0f:a0:00:80:45:20:01:11
16:15:02.164 T:1574278048   DEBUG: CecLogMessage - TV (0): power status changed from 'on' to 'standby'
16:15:02.164 T:1574278048   DEBUG: CecLogMessage - >> TV (0) -> Broadcast (F): vendor command with id (A0)
16:15:02.234 T:1574278048   DEBUG: CecLogMessage - >> 0f:36
16:15:02.234 T:1574278048   DEBUG: CecLogMessage - >> TV (0) -> Broadcast (F): standby (36)
```
The TV is powered off here, not initiated by Kodi.

```
16:15:02.251 T:1942373280   DEBUG: CAnnouncementManager - Announcement: OnScreensaverDeactivated from xbmc
16:15:02.251 T:1942373280   DEBUG: GOT ANNOUNCEMENT, type: 4, from xbmc, message OnScreensaverDeactivated
16:15:02.255 T:1574278048   DEBUG: CecLogMessage - Playback 1 (4) was already marked as active source
16:15:02.255 T:1574278048   DEBUG: CecLogMessage - TV (0): power status changed from 'standby' to 'in transition from standby to on'
16:15:02.255 T:1574278048   DEBUG: CecLogMessage - >> source activated: Playback 1 (4)
16:15:02.255 T:1574278048   DEBUG: CecLogMessage - sending active source message for 'Playback 1'
16:15:02.255 T:1574278048   DEBUG: CecLogMessage - << powering on 'TV' (0)
16:15:02.255 T:1574278048   DEBUG: CecLogMessage - << 40:04
```
Kodi deactivates the screensaver as a result, and the timestamps suggest that it's a result from the TV powering off, which then re-activates the TV.

I don't see libCEC doing anything bad here, but it looks like a bug on Kodi's end, which should be registered on Kodi's Trac instead. There also hasn't been any (released) libCEC update recently. I've got another update almost ready, but I won't have time to finish this before next week because of other priorities unfortunately.

Closing this ticket. Feel free to reopen it if it's a problem in libCEC itself after all.

closed 2017-08-28 by opdenkamp