Pulse-Eight/libcec #365: Do not install libcec - install cec-utils instead!

OZ1SEJ, 2017-07-31:
I have spent two hours trying to git clone, download, compile and I don't know what libcec, only to find out - by chance, on another website - that the command cec-client is contained within something called cec-utils. This isn't mentioned anywhere, so to save others the hazzle, use

`sudo apt-get install cec-utils`

opdenkamp, 2017-08-01:
`libcec` is the library, `cec-utils` contains the utilities (including `cec-client` which is a debugging client for `libcec`) and it will automatically install `libcec` as dependency or it won't work. It's been like this for over 5 years.

closed 2017-08-01 by opdenkamp

OZ1SEJ, 2017-08-01:
Yeah, I found out - after a while. The trouble is, I, and probably a lot of other people, are having trouble getting their remote controls to work with Kodi. And when you google that problem, a lot of people are giving the advice to run "cec-client". When you do that, of course, you're told that cec-client is not installed. The next logical step is to try "sudo apt-get install cec-client" which - I now know why - doesn't work.

There is no mention anywhere that cec-client is part of a larger program called cec-utils - you're only told that you need to git clone and compile libcec which is quite time consuming if you're not used to it. It also turned out to be not necessary.

Is there an established procedure for finding out which installable program a desired command is part of?

opdenkamp, 2017-08-01:
Kodi runs with `libcec` only, and you don't need cec-client to debug problems with Kodi and CEC. Just enable debug logging in Kodi (debug logging, component specific logging, logging for libCEC in system -> debugging). That will give you the exact same output as cec-client, in the correct context, and including the resulting actions in Kodi. cec-client is only useful for some very specifc things, if you know what you're doing.

`apt-cache search` can be used to search for packages.

OZ1SEJ, 2017-08-01:
Thank you, I didn't know that command :-) Now, if I write `apt-cache search cec-client` I get nothing. If I just write `apt-cache search cec` I get hundreds of packages. How would I know which one is the right one?

Interesting about the logging possibilities - I look forward to try it out when I get home from work ;-)

opdenkamp, 2017-08-01:
Try `apt-cache search libcec` and you'll get a list of all our packages.

OZ1SEJ, 2017-08-01:
Ah, I see. So - what you're saying is that there is no "official" way to find out what package some command belongs to - you have to do some digging?

opdenkamp, 2017-08-01:
Debian(-based) is not the only distribution out there, so we can't really provide any "official" way for every distribution. That's why we just describe how to compile the package from source in the readme file https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md

However, because lots of people who use Kodi also use Ubuntu, we do provide builds for this distribution: https://launchpad.net/~pulse-eight/+archive/ubuntu/libcec

You can see a list of packages on that page. Debian have included these in their distribution: https://packages.debian.org/source/sid/libcec

These are "official" ways to find out which package provides what for your distribution, but it doesn't apply to all Linux users.

OZ1SEJ, 2017-08-01:
Interesting! Thank you very much for the explanation! :-)

taa1, 2019-01-12:
@OZ1SEJ 

I believe the program you're looking for is:

`apt-get install apt-file`

Then you can do things like this:

`$ apt-file search cec-client`

which gives what you're looking for:

`cec-utils: /usr/bin/cec-client`

`cec-utils: /usr/bin/cec-client-4.0.2`

`cec-utils: /usr/share/man/man1/cec-client.1.gz`

Note that `apt-file` doesn't access the internet. It relies on the local apt database so results are only as current as the last time you've run `apt-get update`.